lxi-tools/lxi-tools #1: Make Error

HoracioDos, 2017-10-07:
Hello!
I get this error when I try to compile lxi-tools
`Making all in src
make[1]: Entering directory '/home/horacio/Documents/Electronics/DS1054Z/lxi-tools-1.0/src'
  CC       lxi-lxi.o
  CC       lxi-options.o
  CCLD     lxi
  CC       rigol_1000z_screenshot.o
  CCLD     rigol_1000z_screenshot
make[1]: *** No rule to make target 'bash-completion/lxi', needed by 'all-am'.  Stop.
make[1]: Leaving directory '/home/horacio/Documents/Electronics/DS1054Z/lxi-tools-1.0/src'
Makefile:391: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
`
Thanks in advance

lundmar, 2017-10-07:
Hi there,
I need more details - could you please provide all the commands you performed to get to this error?
Also, did you use the officially released .tar.xz file?
Thanks.

HoracioDos, 2017-10-07:
Yes, I downloaded the official release from here:
https://github.com/lxi/lxi-tools/releases/download/v1.0/lxi-tools-1.0.tar.xz
Then I untared the file into a working folder and ran ./configure. This is the output.
[configure output.txt](https://github.com/lxi/lxi-tools/files/1366456/configure.output.txt)
Next I ran make

lundmar, 2017-10-07:
Thanks for the quick input.

I have fixed the bug in commit d04ba0a1db698f8844567f76673d84ccae42f82f and made a new lxi-tools v1.1 release that includes the fix.

HoracioDos, 2017-10-07:
It compiles as expected!
Thanks!!

closed 2017-10-07 by HoracioDos