lxi-tools/lxi-tools #6: ./configure

3hristian, 2017-11-23:
Having a hard time following the Manual.
It says in section 3.1.

> $ ./configure
>     $ make
>     $ make install
> 
>     NOTE: lxi-tools depends on liblxi so you need to go install liblxi first.

Yet without a configure script, it will not budge.
I know I might have missed a trick here but anyhow would be sweet if it works just like in the reedmee.

lundmar, 2017-11-23:
I'm sorry. I know it is a bit generic. The install steps for source assumes that the user is familiar the whole configure/make install thing and how to tailor it for their specific install environment. In other words, the specific configure/make steps will vary depending on which system you are installing and to which install location you want to install etc..

However, you can find a detailed description on how to install from git here:
https://www.eevblog.com/forum/testgear/open-source-lxi-tools-and-liblxi-v1-0-released-for-gnulinux

I hope that helps you.

lundmar, 2017-11-23:
Oh, and by the way. The officially released tarballs always include the configure script.

https://github.com/lxi/lxi-tools/releases/download/v1.10/lxi-tools-1.10.tar.xz
https://github.com/lxi/liblxi/releases/download/v1.5/liblxi-1.5.tar.xz

lundmar, 2017-11-23:
I've updated to READMEs to emphasize the use of release tarballs.

3hristian, 2017-11-24:
Sweet!

closed 2017-11-24 by 3hristian