lxi-tools/lxi-tools #10: Screenshot heterogeneous output

cburlacu, 2017-12-10:
Hi,
The output of screenshot is heterogeneous: bmp, png, jpg. Having multiple instruments it is hard to remember what is the output of each.
I suggest to make it one format (or at least a switch to choose) or either to send the output directly to std in order to be used with imagemagick.
Thanks

lundmar, 2017-12-10:
Hi,
It is recommended that the screenshot plugins use the lossless PNG format if possible but in some cases retrieving a PNG takes significantly longer compared to BMP (e.g. 7s vs 14s). It is slower because the instrument chipset is slow to compress the image data. In other cases the instrument only supports a single format such as BMP. Therefore we let the plugin decide which format is best to use for their particular instrument and that is why the output format might differ depending on which instrument is in play.

However, you can actually do what your suggest already by specifying the output file so it can be used with imagemagick for automatic conversion. For example:

 $ lxi screenshot --address 10.0.0.42 screenshot && convert screenshot screenshot.png

I'm not too keen on the idea of letting the lxi tool automatically do the image conversion because it will introduce a dependency on e.g. the imagemagick library. I think it is best to simply provide the original image directly from the instruments and then let the user convert it to whatever format is needed.

That being said, if enough users really want this automatic conversion, I might give in 😃

cburlacu, 2017-12-10:
Could it be:
```lxi screenshot --address 10.0.0.42 - | convert - screenshot.png``` ?
In this way it won't create a temporary file...

lundmar, 2017-12-10:
Sure. Added by commit 759a7595825b9f2ffd8faa462e1b0c1004a53365

If you are using snap it should be available shortly in the --edge channel:

$ snap install lxi-tools --edge

Or if you already have it installed:

$ snap refresh lxi-tools --edge

Let me know if it works for you.

P.s. In case you are interested there is an active discussion thread on lxi-tools [here](https://www.eevblog.com/forum/testgear/open-source-lxi-tools-and-liblxi-v1-0-released-for-gnulinux/?all)

cburlacu, 2017-12-10:
It's working. Thanks for such prompt response and fix.
Good job!

closed 2017-12-10 by cburlacu

lundmar, 2017-12-10:
You welcome. And thank you for pointing it out 👍 

I'm glad we sorted out this little but useful detail. Documentation has also be updated.