lxi-tools/lxi-tools #12: configure fails to find QT libs on FreeBSD

dmgk, 2018-02-07:
v1.18 switched from PKG_CHECK_MODULES to AC_CHECK_LIB, which broke FreeBSD build:

```
(cd /usr/home/dg/ports/misc/lxi-tools/work/lxi-tools-1.18 &&  _LATE_CONFIGURE_ARGS="" ;  if [ -z "" ] &&  ./configure --help 2>&1 | /usr/bin/grep -- --localstatedir > /dev/null; then  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} --localstatedir=/var" ;  fi ;  if [ ! -z "`./configure --help 2>&1 | /usr/bin/grep -- '--mandir'`" ]; then  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} --mandir=/usr/local/man" ;  fi ;  if [ ! -z "`./configure --help 2>&1 | /usr/bin/grep -- '--disable-silent-rules'`" ]; then  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} --disable-silent-rules" ;  fi ;  if [ ! -z "`./configure --help 2>&1 | /usr/bin/grep -- '--enable-jobserver\[.*#\]'`" ]; then  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} --enable-jobserver=4" ;  fi ;  if [ ! -z "`./configure --help 2>&1 | /usr/bin/grep -- '--infodir'`" ]; then  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} --infodir=/usr/local/info/" ;  fi ;  if [ -z "`./configure --version 2>&1 | /usr/bin/egrep -i '(autoconf.*2\.13|Unrecognized option)'`" ]; then  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} --build=amd64-portbld-freebsd11.1" ;  else  _LATE_CONFIGURE_ARGS="${_LATE_CONFIGURE_ARGS} amd64-portbld-freebsd11.1" ;  fi ;  if ! /usr/bin/env CC="cc" CPP="cpp" CXX="c++"  CFLAGS="-O2 -pipe  -D_WITH_GETLINE -fstack-protector -isystem /usr/local/include -fno-strict-aliasing" CPPFLAGS="-I/usr/local/include -isystem /usr/local/include" CXXFLAGS="-O2 -pipe -D_WITH_GETLINE -fstack-protector -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include"  LDFLAGS=" -L/usr/local/lib -fstack-protector" LIBS="-L/usr/local/lib"  INSTALL="/usr/bin/install -c"  INSTALL_DATA="install  -m 0644"  INSTALL_LIB="install  -s -m 0644"  INSTALL_PROGRAM="install  -s -m 555"  INSTALL_SCRIPT="install  -m 555"  QT_SELECT=qt5 QMAKEMODULES="/usr/home/dg/ports/misc/lxi-tools/work/lxi-tools-1.18/mkspecs/modules:/usr/local/lib/qt5/mkspecs/modules" PKG_CONFIG=pkgconf XDG_DATA_HOME=/usr/home/dg/ports/misc/lxi-tools/work  XDG_CONFIG_HOME=/usr/home/dg/ports/misc/lxi-tools/work  HOME=/usr/home/dg/ports/misc/lxi-tools/work PATH=/usr/home/dg/ports/misc/lxi-tools/work/.bin:/home/dg/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin:/home/dg/bin SHELL=/bin/sh CONFIG_SHELL=/bin/sh QTDIR="/usr/local/lib/qt5" QMAKE="/usr/local/lib/qt5/bin/qmake"  MOC="/usr/local/lib/qt5/bin/moc" RCC="/usr/local/lib/qt5/bin/rcc" UIC="/usr/local/lib/qt5/bin/uic"  QMAKESPEC="/usr/local/lib/qt5/mkspecs/freebsd-$(ccver="$(c++ --version)"; case "$ccver" in *clang*) echo clang ;; *) echo g++ ;; esac)" CMAKE_PREFIX_PATH="/usr/local" CONFIG_SITE=/home/dg/ports/Templates/config.site lt_cv_sys_max_cmd_len=262144 ./configure --disable-static --without-bash-completion-dir --enable-lxi-gui --with-qt-includes=/usr/local/include/qt5  --with-qt-libraries=/usr/local/lib/qt5  --with-extra-includes=/usr/local/include  --with-extra-libs=/usr/local/lib --x-libraries=/usr/local/lib --x-includes=/usr/local/include --prefix=/usr/local ${_LATE_CONFIGURE_ARGS}; then  echo "===>  Script \"configure\" failed unexpectedly.";  (echo "Please report the problem to dg@syrec.org [maintainer] and attach the \"/usr/home/dg/ports/misc/lxi-tools/work/lxi-tools-1.18/config.log\" including the output of the failure of your make command. Also, it might be a good idea to provide an overview of all packages installed on your system (e.g. a /usr/local/sbin/pkg-static info -g -Ea).") | /usr/bin/fmt 75 79 ;  false;  fi)
configure: WARNING: unrecognized options: --with-qt-includes, --with-qt-libraries, --with-extra-includes, --with-extra-libs
```

configure.ac expects to find libQt5*.so in "the usual places" but on FreeBSD they are in /usr/local/lib/qt5. Ports framework passes QT include/lib paths in `--with-qt-includes` and `--with-qt-libraries` args which need to be respected for confugure to succeed.

lundmar, 2018-02-07:
Try something like:
``` ./configure --prefix=/usr/local LDFLAGS=-L/usr/local/lib/qt5 --enable-lxi-gui``` and make sure the .so files are actually in that directory.

This should pass the configure checks. After that, qmake does it's own thing to detect the QT5 libraries.

dmgk, 2018-02-07:
This would work as a workaround, my point was that configure should probably respect `--with-qt-...` too as they seem to be the standard way of passing QT paths.

lundmar, 2018-02-08:
I know. I tried adding QT flags but gave it up because it just became too messy because because mixing autotools and qmake is not pretty. I'll see if I can find time to revisit the issue.

lundmar, 2018-02-08:
I've reworked the configure script to use pkg-config again to check for Qt5. This will have to do for now.

dmgk, 2018-02-08:
@lundmar Thanks!

closed 2018-02-08 by lundmar