lxi-tools/lxi-tools #14: Compile Warnings under Fedora 27 and gcc 7.3.1

DwaineGarden, 2018-03-20:
Just reporting some gcc compiler warnings....

make[1]: Entering directory '/usr/src/lxi-tools/src'
  CC       libapp_a-options.o
  CC       libapp_a-discover.o
  CC       libapp_a-scpi.o
  CC       libapp_a-screenshot.o
  CC       libapp_a-benchmark.o
  CC       libapp_a-run.o
  CC       libapp_a-lxilua.o
lxilua.c: In function ‘connect’:
lxilua.c:51:26: warning: passing argument 1 of ‘lxi_connect’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     device = lxi_connect(address, 0, NULL, option.timeout, VXI11);
                          ^~~~~~~
In file included from lxilua.c:36:0:
/usr/local/include/lxi.h:63:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_connect(char *address, int port, char *name, int timeout, lxi_protocol_t protocol);
     ^~~~~~~~~~~
lxilua.c: In function ‘scpi’:
lxilua.c:83:31: warning: passing argument 2 of ‘lxi_send’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     length = lxi_send(device, command, strlen(command), option.timeout);
                               ^~~~~~~
In file included from lxilua.c:36:0:
/usr/local/include/lxi.h:64:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_send(int device, char *message, int length, int timeout);
     ^~~~~~~~
lxilua.c:92:18: warning: passing argument 1 of ‘question’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     if (question(command))
                  ^~~~~~~
lxilua.c:42:12: note: expected ‘char *’ but argument is of type ‘const char *’
 extern int question(char *string);
            ^~~~~~~~
lxilua.c: In function ‘scpi_raw’:
lxilua.c:132:31: warning: passing argument 2 of ‘lxi_send’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     length = lxi_send(device, command, strlen(command), option.timeout);
                               ^~~~~~~
In file included from lxilua.c:36:0:
/usr/local/include/lxi.h:64:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_send(int device, char *message, int length, int timeout);
     ^~~~~~~~
lxilua.c:141:18: warning: passing argument 1 of ‘question’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     if (question(command))
                  ^~~~~~~
lxilua.c:42:12: note: expected ‘char *’ but argument is of type ‘const char *’
 extern int question(char *string);
            ^~~~~~~~
  CC       plugins/libapp_a-screenshot_keysight-dmm.o
  CC       plugins/libapp_a-screenshot_keysight-ivx.o
  CC       plugins/libapp_a-screenshot_rigol-1000z.o
  CC       plugins/libapp_a-screenshot_rigol-2000.o
  CC       plugins/libapp_a-screenshot_rigol-dg4000.o
  CC       plugins/libapp_a-screenshot_rigol-dm3068.o
  CC       plugins/libapp_a-screenshot_rigol-dp800.o
  CC       plugins/libapp_a-screenshot_rigol-dsa.o
  CC       plugins/libapp_a-screenshot_rohde-schwarz-hmo-rtb.o
  CC       plugins/libapp_a-screenshot_siglent-sdm3000.o
  CC       plugins/libapp_a-screenshot_siglent-sdg.o
  CC       plugins/libapp_a-screenshot_siglent-sds.o
  CC       plugins/libapp_a-screenshot_siglent-ssa3000x.o
  CC       plugins/libapp_a-screenshot_tektromake[1]: Entering directory '/usr/src/lxi-tools/src'
  CC       libapp_a-options.o
  CC       libapp_a-discover.o
  CC       libapp_a-scpi.o
  CC       libapp_a-screenshot.o
  CC       libapp_a-benchmark.o
  CC       libapp_a-run.o
  CC       libapp_a-lxilua.o
lxilua.c: In function ‘connect’:
lxilua.c:51:26: warning: passing argument 1 of ‘lxi_connect’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     device = lxi_connect(address, 0, NULL, option.timeout, VXI11);
                          ^~~~~~~
In file included from lxilua.c:36:0:
/usr/local/include/lxi.h:63:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_connect(char *address, int port, char *name, int timeout, lxi_protocol_t protocol);
     ^~~~~~~~~~~
lxilua.c: In function ‘scpi’:
lxilua.c:83:31: warning: passing argument 2 of ‘lxi_send’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     length = lxi_send(device, command, strlen(command), option.timeout);
                               ^~~~~~~
In file included from lxilua.c:36:0:
/usr/local/include/lxi.h:64:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_send(int device, char *message, int length, int timeout);
     ^~~~~~~~
lxilua.c:92:18: warning: passing argument 1 of ‘question’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     if (question(command))
                  ^~~~~~~
lxilua.c:42:12: note: expected ‘char *’ but argument is of type ‘const char *’
 extern int question(char *string);
            ^~~~~~~~
lxilua.c: In function ‘scpi_raw’:
lxilua.c:132:31: warning: passing argument 2 of ‘lxi_send’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     length = lxi_send(device, command, strlen(command), option.timeout);
                               ^~~~~~~
In file included from lxilua.c:36:0:
/usr/local/include/lxi.h:64:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_send(int device, char *message, int length, int timeout);
     ^~~~~~~~
lxilua.c:141:18: warning: passing argument 1 of ‘question’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     if (question(command))
                  ^~~~~~~
lxilua.c:42:12: note: expected ‘char *’ but argument is of type ‘const char *’
 extern int question(char *string);
            ^~~~~~~~
  CC       plugins/libapp_a-screenshot_keysight-dmm.o
  CC       plugins/libapp_a-screenshot_keysight-ivx.o
  CC       plugins/libapp_a-screenshot_rigol-1000z.o
  CC       plugins/libapp_a-screenshot_rigol-2000.o
  CC       plugins/libapp_a-screenshot_rigol-dg4000.o
  CC       plugins/libapp_a-screenshot_rigol-dm3068.o
  CC       plugins/libapp_a-screenshot_rigol-dp800.o
  CC       plugins/libapp_a-screenshot_rigol-dsa.o
  CC       plugins/libapp_a-screenshot_rohde-schwarz-hmo-rtb.o
  CC       plugins/libapp_a-screenshot_siglent-sdm3000.o
  CC       plugins/libapp_a-screenshot_siglent-sdg.o
  CC       plugins/libapp_a-screenshot_siglent-sds.o
  CC       plugins/libapp_a-screenshot_siglent-ssa3000x.o
  CC       plugins/libapp_a-screenshot_tektronix.o
  AR       libapp.a
  CC       main.o
  CCLD     lxi
make[1]: Leaving directory '/usr/src/lxi-tools/src'
nix.o
  AR       libapp.a
  CC       main.o
  CCLD     lxi
make[1]: Leaving directory '/usr/src/lxi-tools/src'

lundmar, 2018-03-20:
Thanks. Yes, it bugs the eyes :eye: They are on my todo list :smile:

HoracioDos, 2018-03-22:
Hello!
Fewer warnings in Mint 18.3
Thanks!
[compile.log](https://github.com/lxi-tools/lxi-tools/files/1837548/compile.log)

lundmar, 2018-03-22:
ok guys, I've fixed the warnings! :monkey_face:

closed 2018-03-22 by lundmar

DwaineGarden, 2018-03-23:
Looks good.   Completely clean build in Fedora 27 and gcc 7.3.1

lundmar, 2018-03-23:
Yes, clean as a whistle now! :smile:

HoracioDos, 2018-03-23:
Hi Martin.
Still warning messages with gcc 5.4.1. and lua5.2 I downloaded master and run 
./autogen.sh 
./configure
make
Thanks again
[configure.log](https://github.com/lxi-tools/lxi-tools/files/1841759/configure.log)
[make.log](https://github.com/lxi-tools/lxi-tools/files/1841760/make.log)
[config.log](https://github.com/lxi-tools/lxi-tools/files/1841773/config.log)

!

HoracioDos, 2018-03-23:
Hello!
I forgot to compile liblxi. It's so obvious now! I'm sorry for the inconvenience.
Thanks!

lundmar, 2018-03-23:
No worries :smile: