lxi-tools/lxi-tools #19: Why is root access required?

MarkSwanson, 2019-05-07:
$ lxi
cannot create user data directory: /home/username/snap/lxi-tools/1052: Not a directory

The directory exists, and the permissions are fine.

$ strace -f lxi
...
write(2, "need to run as root or suid", 27need to run as root or suid) = 27
...
+++ exited with 1 +++

The lxi program runs fine as root.
Is it possible to run lxi without root privileges?

Thanks!

lundmar, 2019-05-07:
You should be able to run as normal user.

Are you running lxi from a network mapped home directory?

Snaps run in a container which comes with some restrictions - for example, it can only write files (screenshot files etc.) to a normal non network mapped home directory. 

If everything fails you could try snap install with the --classic option, this way you bypass some of the container restrictions.

MarkSwanson, 2019-05-08:
I'm not running on a network mapped home dir.

Again, everything works when built from source (well, the GUI doesn't work as there were no Ubuntu packages for the Qt software required). But I can do scpi and take a screenshot :-)

closed 2019-05-08 by MarkSwanson

lundmar, 2019-05-08:
I'm curious, what distribution are you running there?

It is possible that some distributions have a broken or way out of date snapcraft installation.

```
$ snap --version
snap    2.38.1
snapd   2.38.1
series  16
ubuntu  19.04
kernel  5.0.0-13-generic
```

lundmar, 2019-05-08:
Despite snap/snapd being up to date I bet the issue you see is specific for Ubuntu 16.04. I would recommend upgrading to latest Ubuntu version :wink: