lxi-tools/lxi-tools #20: Siglent SDS1202X-E Data Recorder Example SCPI

ecbrown, 2019-05-12:
Hi All,
 
I'm rather new to scopes in general, but have managed to compile lxi-gui for Debian GNU/Linux (buster).  I am able to discover the scope, and get screen captures with no problem.

However, I am trying to figure out how to get the Data Recorder working for the SDS1202X-E.  When I parrot the example and try READ? in Data 1, it crashes the scope and lxi-gui app with a "Failed to receive message!" error. (I don't even think this command is listed in the SCPI PDF manual for this series.)

Does anyone have an example SCPI Command that would work for this instrument so that the Data Recorder works as expected?

Thanks!

Eric

Siglent Technologies,SDS1202X-E,SDS1ECDQ2R5789,7.1.1.3.23

lundmar, 2019-05-13:
Hi Eric,

I don't think the READ? command is supported by this instrument. You might want to take a look at the SCPI programming guide for the SDS1000X-E series: https://www.siglentamerica.com/wp-content/uploads/dlm_uploads/2017/10/Programming-Guide-1.pdf

Last time I checked I didn't find any easy way to read out e.g. the voltage of a channel signal.

FYI - Siglents video, from which I assume you got the "READ?" hint, is actually testing using a SDM and not a SDS instrument.

ecbrown, 2019-05-13:
Thanks for the comment, Martin.  I have written Siglent with this issue.  Hopefully it can be addressed in a firmware update.  Very important IMO for a scope to read & report voltage on the channels!

ecbrown, 2019-05-14:
Hi Martin,

Got a nice email from Jason at Siglent.  He suggested "C1:WF? DAT2" -- which I had discovered before, but turned away due to its binary nature, hoping to find something ASCII.  (I think it's US-ASCII once decoded.)

There is a Kotlin example of decoding is toward the bottom of this:
[Kotlin SDS1202X-E Waveform](https://www.siglentamerica.com/application-note/x-e-kotlin-example/)

Wondering whether the  whole waveform is a unique task vis-a-vis lxi-tools and the Data Recorder, compared to e.g. a single voltage being returned.

So, it looks pretty simple.  I might get the chance to hack something in to lxi-tools, but if anyone beats me to it, then by all means. :-)

I've attached the outputs of:
a) the gzipped output of  "C1:WF? DAT1" as output1.bin.gz
b) the gzipped output of  "C1:WF? DAT2" as output2.bin.gz
c) the Kotlin source (BSD license)

Best regards,
Eric

[output1.bin.gz](https://github.com/lxi-tools/lxi-tools/files/3175302/output1.bin.gz)
[output2.bin.gz](https://github.com/lxi-tools/lxi-tools/files/3175303/output2.bin.gz)
[SDSDataRetrievalKotlinExample.txt](https://github.com/lxi-tools/lxi-tools/files/3175304/SDSDataRetrievalKotlinExample.txt)

lundmar, 2019-05-14:
Yeah, unfortunately the Siglent SDS instrument does not include a waveform statistical analysis feature so it requires a bit cumbersome post processing to get basic metrics from a sampled waveform.

I hope Siglent will include a statistics feature in their future firmware updates or next gen oscilloscope. Rigol 1000 series is a good example of how to implement the statistics feature - for example to query the DSO voltage RMS one simply do ":MEASure:STATistic:ITEM? current,vrms" and the voltage value is returned in standard scientific notation which is supported by the lxi-gui data recorder application. Simple.

ecbrown, 2019-05-26:
Should probably report back to this issue: I think most of my "issues" can be triggered by having Men Depth too large on the scope: Acquire... Mem Depth... set to 14K and everything is fine with waveform export, including lxi-tools.

closed 2019-05-26 by ecbrown

torquil, 2019-08-21:
Hi Eric! I'm also using lxi-tools together with a similar scope, a Siglent SDS1104X-E. But I'm getting lots of problems with obtaining the waveform when the memory depth is at the maximum value. This is also the case with my attempts at using pyvisa with the pyvisa-py backend. The problems when using lxi-tools involve segfaults and the scope freezing up. With pyvisa I'm getting lots of problems with truncated output when reading from the scope. I'm using the newest firmwave, 6.1.33.

Did you manage to find a "fool-proof" way to extract the full waveform using lxi-tools when the memory depth is set to the highest values?

Best regards,
Torquil Sørensen

wijjj, 2020-04-23:
I'm also struggling with reading from the SDS 1202x-e. Anybody made any progress yet?