lxi-tools/lxi-tools #21: Building lxi-gui on archlinux fails

nickoe, 2019-05-23:
I get qmake command not found from make, even though I do have qmake on Archlinux. It also looks like configure did find qmake-qt5...

```
[nickoe@z620 lxi-tools-git]$ ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:6: installing './compile'
configure.ac:4: installing './missing'
src/Makefile.am: installing './depcomp'
autoreconf: Leaving directory `.'
[nickoe@z620 lxi-tools-git]$ ./configure --prefix=/usr --enable-lxi-gui
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for readline in -lreadline... yes
checking for lxi_connect in -llxi... yes
checking for lua... no
checking for lua... no
checking for lua... no
checking for lua... yes
checking for moc-qt5... moc-qt5
checking for uic-qt5... uic-qt5
checking for rcc... rcc
checking for qmake-qt5... qmake-qt5
checking for qtchooser... no
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/gui/lxi-gui/Makefile
config.status: creating man/Makefile
config.status: creating src/include/config.h
config.status: src/include/config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
[nickoe@z620 lxi-tools-git]$ qmake --version
QMake version 3.1
Using Qt version 5.12.3 in /usr/lib
[nickoe@z620 lxi-tools-git]$ make
Making all in src
make[1]: Entering directory '/tmp/lxi-tools-git/src/lxi-tools-git/src'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/tmp/lxi-tools-git/src/lxi-tools-git/src'
Making all in man
make[1]: Entering directory '/tmp/lxi-tools-git/src/lxi-tools-git/man'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/tmp/lxi-tools-git/src/lxi-tools-git/man'
Making all in src/gui/lxi-gui
make[1]: Entering directory '/tmp/lxi-tools-git/src/lxi-tools-git/src/gui/lxi-gui'
run-tool=qmake -qt=5 -makefile -o Makefile.qmake lxi-gui.pro \
	QT_INCDIR="//usr/include" \
	QT_LIBDIR="//usr/lib" \
	SNAPCRAFT="" \
	
/bin/sh: run-tool=qmake: command not found
make[1]: [Makefile:466: Makefile.qmake] Error 127 (ignored)
make -f Makefile.qmake  all
make[2]: Entering directory '/tmp/lxi-tools-git/src/lxi-tools-git/src/gui/lxi-gui'
make[2]: Makefile.qmake: No such file or directory
make[2]: *** No rule to make target 'Makefile.qmake'.  Stop.
make[2]: Leaving directory '/tmp/lxi-tools-git/src/lxi-tools-git/src/gui/lxi-gui'
make[1]: *** [Makefile:473: all-local] Error 2
make[1]: Leaving directory '/tmp/lxi-tools-git/src/lxi-tools-git/src/gui/lxi-gui'
make: *** [Makefile:408: all-recursive] Error 1
[nickoe@z620 lxi-tools-git]$ 
```

lundmar, 2019-05-23:
It can't find the qtchooser tool which is used to automatically resolve the name of the qmake qt5 binary. If you install qtchooser from AUR I expect it will work. The GUI is beta and apparently so is the configure script :smirk: If qtchooser does not work for you I will find a way to fix the configure script.

nickoe, 2019-05-24:
Ok, I see. I am not very familiar with the qt build system. I guess I will just add qtchooser as a dependency to my package as well.

fantomsam, 2019-06-02:
I had some similar problem and I solve it like so. 
Try to install qt5-qmake and qt5-default
Clone the source.
Browse in to the gui directory inside to the src directory and modify the lxi-gui.pro by commenting the -L"$$QT_LIBDIR" on line 29, save and try to compile the lxi-gui by run make on the gui directory.

nickoe, 2019-06-05:
If the configure script says `checking for qtchooser... no`, maybe the script should just use qmake-qt5 directly from the normal PATH environment?

fantomsam, 2019-06-07:
Try to do the following 
Install QT on your system.
cd on lxi-gui directory  
run qmake from the location that is hve been installed eg. Qt/5.12.3/gcc_64/bin/qmake 
if you don't get any error do make
if you get a "/usr/bin/ld: cannot find -lQt5Charts" error do the modification on the qmake file (lxi-gui.pro) that I have suggest above. and run make again

nickoe, 2019-06-11:
If I do like below, it will build, but it is of course a hack. qtchooser does not seem to be easily available on archlinux at the moment.
```
diff --git a/src/gui/lxi-gui/Makefile.am b/src/gui/lxi-gui/Makefile.am
index 6d0f375..18f8e50 100644
--- a/src/gui/lxi-gui/Makefile.am
+++ b/src/gui/lxi-gui/Makefile.am
@@ -13,7 +13,7 @@ EXTRA_DIST = lxi-gui.pro \
 QT_SELECT ?= 5
 
 Makefile.qmake: lxi-gui.pro
-       $(QTCHOOSER) -run-tool=qmake -qt=$(QT_SELECT) -makefile -o Makefile.qmake lxi-gui.pro \
+       qmake -makefile -o Makefile.qmake lxi-gui.pro \
                QT_INCDIR="$(SNAPCRAFT_STAGE)/$(includedir)" \
                QT_LIBDIR="$(SNAPCRAFT_STAGE)/$(libdir)" \
                SNAPCRAFT="$(SNAPCRAFT)" \
```

peterbabic, 2021-02-19:
There seems to be no packages qt5-qmake and qt5-default. I was able to install qt5-charts and it let me finish the `./configure`. I do not know how to run `make` without error, even after applying the patch proposed by @nickoe 

```
Making all in src
make[1]: Entering directory '/home/peterbabic/.cache/yay/lxi-tools-git/src/lxi-tools-git/src'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/peterbabic/.cache/yay/lxi-tools-git/src/lxi-tools-git/src'
Making all in man
make[1]: Entering directory '/home/peterbabic/.cache/yay/lxi-tools-git/src/lxi-tools-git/man'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/peterbabic/.cache/yay/lxi-tools-git/src/lxi-tools-git/man'
Making all in src/gui/lxi-gui
make[1]: Entering directory '/home/peterbabic/.cache/yay/lxi-tools-git/src/lxi-tools-git/src/gui/lxi-gui'
Makefile:466: *** missing separator.  Stop.
make[1]: Leaving directory '/home/peterbabic/.cache/yay/lxi-tools-git/src/lxi-tools-git/src/gui/lxi-gui'
make: *** [Makefile:410: all-recursive] Error 1
```

hraftery, 2021-03-23:
Thanks @nickoe your diff did the trick on macos 11.2.3 after installing Qt 5 with Homebrew (`brew install qt5`). I also had installed Qt 5 using the online installer, but it seems to want to be self contained so resorted to Homebrew to get some re-usable elements. I still had to follow the advice from Homebrew and manually add the paths:

```
export PATH="/usr/local/opt/qt@5/bin:$PATH"
export LDFLAGS="-L/usr/local/opt/qt@5/lib"
export CPPFLAGS="-I/usr/local/opt/qt@5/include"
export PKG_CONFIG_PATH="/usr/local/opt/qt@5/lib/pkgconfig"
./configure --enable-lxi-gui
make
```

`configure` and `make` were then successful, although `make install` failed - it seemed to be tricked by the fact that `make` correctly created a .app package! Anyway, I could run the app with a double-click. It worked, although crashed after clicking the "live view" button.

This is really coming along. Looking forward to seeing more FOSS support of LXI.

lundmar, 2021-12-25:
Closing issue. Next release will replace autotools with meson and the lxi-gui has been fully rewritten in GTK4.

closed 2021-12-25 by lundmar

nickoe, 2021-12-26:
Ok. I also updated the Archlnux AUR package to use meson such that it builds again.
https://aur.archlinux.org/packages/lxi-tools-git/

lundmar, 2021-12-26:
Great, thanks! 👍🏻