lxi-tools/lxi-tools #22: SCPI simple text script does not work

fantomsam, 2019-06-02:
Hello. I try to test the lxi send script functionality and I found that does not work, I get an error of "No SCPI command specified". By looking the source code I found that on file option.c, in function parse_options on line 404, At the option check does not happens any check for the option.run_script flag and this cause the program to send this message on the case of text script run. 

I attach a patch file with the change that i make.
[options.patch.tar.gz](https://github.com/lxi-tools/lxi-tools/files/3244701/options.patch.tar.gz)

BR
Sam

.

lundmar, 2021-12-25:
Hi Sam. I'm sorry for the very late response but sometimes life gets in the way of maintaining my open source projects. Change looks solid. Would you mind create a pull request instead of a patch? Thanks.

lundmar, 2021-12-25:
I just now see your PR and that it also includes your fix. Good work. I'm closing this issue. I will work on getting your PR included.

closed 2021-12-25 by lundmar