lxi-tools/lxi-tools #24: Can't read nor record data using SDS1104X-E

moxo26, 2019-06-14:
I am trying to use lxi and lxi tools to read data and log it. But I am unable to do it. I am able to live view and screen shot but I can't log the data. 
I installed from:
snap install lxi-tools

When I start the lxi-gui i get the following error:
Gtk-Message: Failed to load module "gail"
Gtk-Message: Failed to load module "atk-bridge"
Qt: Session management error: None of the authentication protocols specified are supported

Furthermore, when I send the SCPI message READ? , I get the following message: -113,"Undefined header;READ?"

cbake6807, 2021-07-25:
@moxo26 Ever figure this out? I'm trying the same, but over cli. READ? just times out any way i try to run it. Same on my SDS 1052DL+

pi@raspberrypi:~ $ lxi scpi --address 192.168.1.7 "*IDN?"
Siglent Technologies,SDS1104X-E,SDSMMFCD4R7397,8.1.6.1.35R2

pi@raspberrypi:~ $ lxi scpi --address 192.168.1.7 "*READ?"
Error: Read error (timeout)
Error: Failed to receive message

lundmar, 2021-12-25:
Guys, you can't just fire any arbitrary SCPI command and expect it to work. According to the SDS1000X-E programming guide there is no support for any "*READ?" command. This is why you get a timeout.

Programming guide:
https://siglentna.com/wp-content/uploads/dlm_uploads/2021/01/SDS1000-SeriesSDS2000XSDS2000X-E_ProgrammingGuide_PG01-E02D.pdf

closed 2021-12-25 by lundmar