lxi-tools/lxi-tools #25: liblxi not found even though it's installed

nonchip, 2019-07-16:
I built & installed liblxi (to /usr/local/lib, which is in my ld.so.conf), but when I run `configure` on lxi-tools I get:

    checking for lxi_connect in -llxi... no
    configure: error: liblxi not found

the linker setup definitely worked though:

    $ ldconfig -p | grep lxi
         liblxi.so.1 (libc6,x86-64) => /usr/local/lib/liblxi.so.1
            liblxi.so (libc6,x86-64) => /usr/local/lib/liblxi.so

so i wonder if `configure` could be trying to override `LD_{LIBRARY,RUN}_PATH` or trying to somehow find the library using some heuristic that's excluding `/usr/local` even though it's a pretty standard path...

there also doesn't seem to be a `--with-lxi=path` parameter for it?

lornix, 2020-12-28:
Be sure to tell 'configure' where the libraries are located.  /lusr/local/lib is a non-default path.

./configure --libdir=/usr/local/lib

--OR--

LDFLAGS=-L/usr/local/lib ./configure

First method is easier.  Basically, linker can see the library (ldconfig -p example), but 'configure' is kinky and wants to *touch* it to verify it exists.

lundmar, 2021-12-25:
Seems you have the liblxi library file but for some reason it does not contain the lxi_init function.

Anyway, a lot of changes are coming, including switching from autotools to meson so you may get a better experience on your distribution soon.

closed 2021-12-25 by lundmar

dwarning, 2022-01-17:
Sorry to re-open this issue.
I tried to build from source on Ubuntu 20.04 and got this:
`Found pkg-config: /usr/bin/pkg-config (0.29.1)
Run-time dependency readline found: YES 8.0
Run-time dependency lua-5.3 found: YES 5.3.3
Found CMake: /usr/bin/cmake (3.16.3)
Run-time dependency liblxi found: NO (tried pkgconfig and cmake)

src/meson.build:42:0: ERROR: Dependency "liblxi" not found, tried pkgconfig and cmake
`
liblxi and liblxi-dev is installed but:
`dietmar@modsys:~/Projects/lxi-tools$ pkg-config --modversion liblxi
Package liblxi was not found in the pkg-config search path.
Perhaps you should add the directory containing `liblxi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'liblxi' found
`
Seems the installation is not complete. Should I install liblxi from source?

lundmar, 2022-01-17:
You are a bit too early - this is an issue of non-released lxi-tools and liblxi.

We've recently migrated to meson and that primarily rely on pkgconfig for resolving library configurations etc. The old liblxi using autotools does not install a pkgconfig. The latest git versions, liblxi installs a pkgconfig configuration so that lxi-tools can find it.

In, in short, install the latest git version of liblxi and it will match.

dwarning, 2022-01-17:
OK, I will wait.
Thank you for the nice work.

lundmar, 2022-01-17:
No problem. It will be released soon. Until then you could try out the edge snap for easy install.

dwarning, 2022-01-17:
I tried snap - but I got
`dietmar@modsys:~$ lxi-gui
Gtk-Message: Failed to load module "gail"
Gtk-Message: Failed to load module "atk-bridge"
Qt: Session management error: Could not open network socket
`
libatk-adaptor libgail-common was installed properly.

lxi-gui seems functiona, but in the end I removed the snap.

lundmar, 2022-01-17:
Since lxi-tools v2.0 has not yet been officially released you need to install the bleeding edge snap like so:

`$ snap install lxi-tools --edge`

nonchip, 2022-01-18:
gotta love the necromancy spam...