lxi-tools/lxi-tools #26: Device not found: LDH400

fantilator, 2019-09-18:
Hi, I have a LDH400 DC load with LXI enabled, still the tools are unable to find the device.

I have tried all mDNS services. 
The linux is running on a VM in the same network as the DC load.
trying the command line give ma also an error.

Can you help me out where it gets stuck?
Kind Regards
Davy

[image]

Please find attached some screenshots.
[image]
[image]

jstiefel, 2020-08-24:
Same problem with the Aim TTi CPX400SP. Connected it to a local switch with a fixed IP and I can ping it, go to the web interface etc.

> EDIT: When rinning the command with -r and specific port it works.
> I cannot change the port number.

This solved it, although I don't understand why a search via mDNS is necessary if I use an IP.

lundmar, 2021-12-25:
To me it seems like your Aim instruments do not support the standard VXI11 protocol but only RAW TCP for sending commands.

lundmar, 2021-12-25:
> This solved it, although I don't understand why a search via mDNS is necessary if I use an IP.
Apparently your instrument does not support the old way of discovery that most instruments support. Maybe Aim decided to leave it behind.

lundmar, 2021-12-25:
> EDIT: When rinning the command with -r and specific port it works. I cannot change the port number. Is it possible to add the feature in the GUI?

The lxi-gui application has recently been fully rewritten in GTK4. I may add such option in the future but the gui is really only meant to be simple and support instruments which adhere strictly to the LXI standard.

lundmar, 2021-12-25:
Oh, by the way. If you guys run stuff in a VM, you have to remember to bridge the VM network with your local network else the discovery broadcast messages will be lost. This may be another reason the discover feature did not work for you.

closed 2021-12-25 by lundmar