lxi-tools/lxi-tools #30: lxi-gui not getting built

SheetLightning, 2021-03-30:
I downloaded the tar archive and tried to compile. The configure script ran fine and identified that lxi-lib needed installing which I did. The configure then completed with no issues. I then ran make. 

Make did produce some warnings as can be seen in the output below, but did complete. The lxi command line tools has been compiled and seems to work fine, but not lxi-gui.

If I navigate to ./src/gui/lxi-gui and run make in there, I get the following:

```
$ make
run-tool=qmake -qt=5 -makefile -o Makefile.qmake lxi-gui.pro \
	QT_INCDIR="//usr/local/include" \
	QT_LIBDIR="//usr/local/lib" \
	SNAPCRAFT="" \
	
/bin/bash: run-tool=qmake: command not found
Makefile:464: recipe for target 'Makefile.qmake' failed
make: [Makefile.qmake] Error 127 (ignored)
make -f Makefile.qmake  all
make[1]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui'
make[1]: Makefile.qmake: No such file or directory
make[1]: *** No rule to make target 'Makefile.qmake'. Stop.
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui'
Makefile:471: recipe for target 'all-local' failed
make: *** [all-local] Error 2
```

Qmake is installed. After running make a couple more times I got:

```
$ make
/usr/lib/x86_64-linux-gnu/qt4/bin/uic mainwindow.ui -o ui_mainwindow.h
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I. -I. -o main.o main.cpp
In file included from main.cpp:1:0:
mainwindow.h:6:10: fatal error: QtCharts/QChartView: No such file or directory
 #include <QtCharts/QChartView>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:231: recipe for target 'main.o' failed
make: *** [main.o] Error 1
```

When I run 'qmake' I get:

```
$ qmake
Project MESSAGE: Warning: unknown QT: charts
```

So it seems that QT Charts is missing, but where do i install it from? I couln't find it in the Mint repository nor do i find anything useful online?
Not sure what the problem with qmake is either as it is definitely installed on the system?

Output of configure:
```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
[... 150 lines cut ...]
  CC       plugins/libapp_a-screenshot_rigol-2000.o
  CC       plugins/libapp_a-screenshot_rigol-dg4000.o
  CC       plugins/libapp_a-screenshot_rigol-dm3068.o
  CC       plugins/libapp_a-screenshot_rigol-dp800.o
  CC       plugins/libapp_a-screenshot_rigol-dsa.o
  CC       plugins/libapp_a-screenshot_rohde-schwarz-hmo-rtb.o
  CC       plugins/libapp_a-screenshot_siglent-sdm3000.o
  CC       plugins/libapp_a-screenshot_siglent-sdg.o
  CC       plugins/libapp_a-screenshot_siglent-sds.o
  CC       plugins/libapp_a-screenshot_siglent-ssa3000x.o
  CC       plugins/libapp_a-screenshot_tektronix.o
  AR       libapp.a
ar: `u' modifier ignored since `D' is the default (see `U')
  CC       main.o
  CCLD     lxi
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src'
Making all in man
make[1]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/man'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/man'
make[1]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21'
johnc@agabus ~/src/lxi-tools/lxi-tools-1.21 $ 
```

SheetLightning, 2021-03-30:
With a bit of help I made some progress.

It seems that to compile the gui one needs to run configure with the  --enable-lxi-gui flag

`./configure --enable-lxi-gui
`
Configure then reports that it needs QtChart. I was informed that the library to install is libqt5charts5-dev. After that, configure succeeds and make attempts to compile the gui. However, it then fails with errors which seem to relate to the previous warnings.

```
$ make
Making all in src
make[1]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src'
  CC       libapp_a-options.o
  CC       libapp_a-discover.o
discover.c: In function ‘discover’:
discover.c:66:20: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     info.broadcast = &broadcast;
                    ^
discover.c:67:17: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     info.device = &device;
                 ^
discover.c:68:18: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     info.service = &service;
                  ^
  CC       libapp_a-scpi.o
  CC       libapp_a-screenshot.o
  CC       libapp_a-benchmark.o
  CC       libapp_a-run.o
  CC       libapp_a-lxilua.o
lxilua.c: In function ‘connect’:
lxilua.c:51:26: warning: passing argument 1 of ‘lxi_connect’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     device = lxi_connect(address, 0, NULL, option.timeout, VXI11);
                          ^~~~~~~
In file included from lxilua.c:36:0:
/usr/include/lxi.h:59:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_connect(char *address, int port, char *name, int timeout, lxi_protocol_t protocol);
     ^~~~~~~~~~~
lxilua.c: In function ‘scpi’:
lxilua.c:83:31: warning: passing argument 2 of ‘lxi_send’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     length = lxi_send(device, command, strlen(command), option.timeout);
                               ^~~~~~~
In file included from lxilua.c:36:0:
/usr/include/lxi.h:60:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_send(int device, char *message, int length, int timeout);
     ^~~~~~~~
lxilua.c: In function ‘scpi_raw’:
lxilua.c:132:31: warning: passing argument 2 of ‘lxi_send’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
     length = lxi_send(device, command, strlen(command), option.timeout);
                               ^~~~~~~
In file included from lxilua.c:36:0:
/usr/include/lxi.h:60:5: note: expected ‘char *’ but argument is of type ‘const char *’
 int lxi_send(int device, char *message, int length, int timeout);
     ^~~~~~~~
  CC       plugins/libapp_a-screenshot_keysight-dmm.o
  CC       plugins/libapp_a-screenshot_keysight-ivx.o
  CC       plugins/libapp_a-screenshot_rigol-1000z.o
  CC       plugins/libapp_a-screenshot_rigol-2000.o
  CC       plugins/libapp_a-screenshot_rigol-dg4000.o
  CC       plugins/libapp_a-screenshot_rigol-dm3068.o
  CC       plugins/libapp_a-screenshot_rigol-dp800.o
  CC       plugins/libapp_a-screenshot_rigol-dsa.o
  CC       plugins/libapp_a-screenshot_rohde-schwarz-hmo-rtb.o
  CC       plugins/libapp_a-screenshot_siglent-sdm3000.o
  CC       plugins/libapp_a-screenshot_siglent-sdg.o
  CC       plugins/libapp_a-screenshot_siglent-sds.o
  CC       plugins/libapp_a-screenshot_siglent-ssa3000x.o
  CC       plugins/libapp_a-screenshot_tektronix.o
  AR       libapp.a
ar: `u' modifier ignored since `D' is the default (see `U')
  CC       main.o
  CCLD     lxi
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src'
Making all in man
make[1]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/man'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/man'
Making all in src/gui/lxi-gui
make[1]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui'
qtchooser -run-tool=qmake -qt=5 -makefile -o Makefile.qmake lxi-gui.pro \
	QT_INCDIR="//usr/local/include" \
	QT_LIBDIR="//usr/local/lib" \
	SNAPCRAFT="" \
	
Info: creating stash file /home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui/.qmake.stash
make -f Makefile.qmake  all
make[2]: Entering directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui'
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I. -isystem /usr/local/include -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCharts -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o main.o main.cpp
main.cpp: In function ‘void lxi_setup()’:
main.cpp:32:26: error: invalid conversion from ‘void (*)(const char*, const char*)’ to ‘void (*)(char*, char*)’ [-fpermissive]
     lxi_info.broadcast = &broadcast;
                          ^~~~~~~~~~
main.cpp:33:23: error: invalid conversion from ‘void (*)(const char*, const char*)’ to ‘void (*)(char*, char*)’ [-fpermissive]
     lxi_info.device = &device;
                       ^~~~~~~
Makefile.qmake:426: recipe for target 'main.o' failed
make[2]: *** [main.o] Error 1
make[2]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui'
Makefile:471: recipe for target 'all-local' failed
make[1]: *** [all-local] Error 2
make[1]: Leaving directory '/home/johnc/src/lxi-tools/lxi-tools-1.21/src/gui/lxi-gui'
Makefile:409: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

```

SheetLightning, 2021-03-31:
Make some further progress. It seems that the problem was the older liblxi-1.8 library installed on my system from the Ubuntu repository. I downloaded and compile the current version of the library from the lxi-tools Github and the program then compiled correctly. There were a couple of issues along the way. Firstly, I needed to add the following to lxi-gui.pro:

`QT     += widgets`

This was so that Qt finds QtAction. After the compile had finished, I had to run:

`sudo` ldconfig

This was to allow the system to find the liblxi.so.1 library, otherwise the following error ocurred:

```
$ lxi-gui
lxi-gui: error while loading shared libraries: liblxi.so.1: cannot open shared object file: No such file or directory
```

The lxi-gui program now works and I can communicate with my Rigol DS1054Z.

lundmar, 2021-12-25:
I don't recommend using that of an old Ubuntu distribution, for many reasons. Doing so you will face problems like the ones you had.

Anyway, big changes are coming in the upcoming release. lxi-gui has been fully rewritten in GTK4 to give it a more modern application UI. Also, autotools has been replaced with meson.

closed 2021-12-25 by lundmar