lxi-tools/lxi-tools #33: Tested  Rigol DG1062Z

werwaswarum, 2022-01-09:
Hi,

Signal generator Rigol DG1022Z works for Discover and Scpi functions 

Best regards

lundmar, 2022-01-09:
Thanks for reporting. I'm pretty sure I can make the screenshot feature work too. Let me make a few changes and get back to you.

werwaswarum, 2022-01-09:
Thanks, a screenshot would be helpful!
Maybe this will be helpful:
For now I have the error response "Failure handling image format" in the GUI.

However, the SCPI commands in screenshot_rigol-dg.c should work, at least I could create a screenshot with Python using the same SCPI commands.

lundmar, 2022-01-09:
Yeah, I see we already have a screenshot plugin for this model.

I'm assuming you are using the snap edge version right?

Could you please try use the commandline tool?

$ lxi-tools.lxi screenshot -a <address>

Thanks.

werwaswarum, 2022-01-09:
Yes I'm using the snap edge version
Using the CLI
`lxi-tools.lxi screenshot -a <address>
Loaded rigol-dg screenshot plugin
Saved screenshot image to screenshot_<address>_2022-01-09_19:44:06.bmp
`

I get a screenshot saved to my PC

lundmar, 2022-01-09:
Good, I expected that to work.

The commandline tool basically dumps the raw image data received from the instrument and you can read it with any image viewer. However, in case of the GUI I have to use GTK's image loader to present the image and when I build the snap it is using an old Ubuntu 18.04 LTS platform which contains some buggy image processing libraries that GTK rely on for its loader and they just fail to detect and read the raw BMP data. I though I had solved the issue in the GUI by telling the loader exactly which format the image is but apparently that is not enough.

I'll have to look into maybe rebuilding newer versions of the snap image libraries to make it work. I'll leave this issue open for now until I get it solved before I officially release the new GUI.

Either way, the commandline works so I've your instrument to the READM.md list of tested instruments and I've also added you to the contributors list.

Perhaps I can get you to retest the GUI once I have fixed the issue.

Thanks.

werwaswarum, 2022-01-09:
I will be happy to continue testing as soon as there is something.
Many thanks for the help so far!

lundmar, 2022-01-10:
Okay, I think I have fixed the issue. I had to reconfigure the build configuration to force build the needed image loader plugins.

The fix is included in the latest edge snap (1193)

Please give it a go.

Thanks.

werwaswarum, 2022-01-10:
I tested the last Edge snap (1193) and indeed I can now use the "screen capture" function in the graphical user interface.

So everything is working fine now!

Thanks again!

closed 2022-01-10 by werwaswarum