lxi-tools/lxi-tools #38: Request: colour-coded transcript

hooperbloob, 2022-01-31:
Nice updated UI but I find it a challenge to follow the conversation (who said what) when using multiple instruments.
It'd be nice if the messages could be color-coded per instrument similar to how Wireshark does it - though I think they assign colour by msg type/status.
So, where the instruments are listed on the left panel, users could either accept default colour assignments or pick their own. The background colour for messages on the right would then  match those.

lundmar, 2022-01-31:
Right, I see how that can be an issue when working with multiple instruments at the same time.

While colour encoding seems like a good idea I do not think it should be the primary way of identifying who said what in your transcript, mostly because there are a lot of colour blind users out there. Also, ultimately, if you have a growing number of instruments, it will be increasingly harder to assign unique colours which you can tell apart, even for people with eagle eyes.

I think what we can do is add an option to prefix the IP address to the messages in the SCPI command window. Much in the same way that Wireshark identifies/sorts messages by src/dst IP.

hooperbloob, 2022-02-01:
The default colour palettes could include greyscale ones as well, everyone can distinguish relative intensity levels.

I like the idea of including prefixes. If you provide those in a separate column you could then add some other columns, like timestamps. Then the users could sort msgs/responses by device, time, or message type.

lundmar, 2022-02-01:
I think I will add preferences options to add timestamp, IP, and maybe message type (req/rsp).

However, sorting and putting stuff in columns I'm not to sure about because the tool is not like wireshark where you have to sort massive amounts of packet data. For the SCPI feature, it is more a grab and go tool to quickly play around and poke the features of your test instruments - anything more complex than that really should be done using the script feature. Also, I certainly do not want it to start looking like wireshark or a spreadsheet. Avoiding that will take some effort but maybe some day..

lundmar, 2022-02-02:
Added in 0d1f083032eb86eff9fbc9ab0f08038e4c7d553a so you can now go to preferences and enable prefix of timestamp, IP, type.

You can try it out on the snap edge channel.

hooperbloob, 2022-02-02:
Looks good.  I think the square brackets could be optional and showing the ip for the response is redundant.  
Might be more interesting to show just the elapsed time for the response:

12:34:12:123  10.0.0.32   IDN?
                          +10ms   Agilent 34401 bla bla

lundmar, 2022-02-02:
Right, I went a little bit excessive with the number of brackets. However, I still want brackets to strongly indicate what part is the prefix. So, I've simplified and now only put two brackets like so:

```
[20:59:44:191 192.168.0.107 REQ] *IDN?
[20:59:44:193 192.168.0.107 RSP] Rohde&Schwarz,NGM202,3638.4472k03/101403,03.068 00A8F863604
[21:00:15:349 192.168.0.107 REQ] :SYSTem:VERSion?
[21:00:15:350 192.168.0.107 RSP] 1999.0
```
I know the receive IP is kind of redundant but it serves the important purpose of aligning the input and out strings - otherwise it would look messy as the request and response strings would constantly zig zag.

I do like the idea of printing the response time but not so much from which instrument ID because that will simply become too long of a string. But printing the response time will also introduce a zig zag effect so I think I will do without it. Also, it is fairly easy to tell the response time from the timestamps when everything is aligned like above.

I'm pretty happy with the current look.