lxi-tools/lxi-tools #39: Bugs: mDNS issues in lxi-gui

hooperbloob, 2022-02-02:
I have some instruments that can be discovered with the default settings and another one that needs to have the mDNS flag enabled before it shows up.   However I can't list all of them that way, it is either/or.

Also, I'm unable to communicate with the mDNS instrument when it shows up, the Send button disables, no response.  Its still disabled even after I do a new search and bring back the non-mDNS instruments.

lundmar, 2022-02-02:
Discovering instruments using the one or the other discovery method is going to be a limitation for now. At some point I will improve management of the instrument list and discovery feature and maybe add the option for adding instruments manually but it will take some effort so it will have to wait for now.

The send button not restoring is certainly a bug. I'll fix that.

lundmar, 2022-02-02:
The send button never restoring it's state happens when one uses the TCP/RAW protocol and connects to a port not supported by the instrument. It should time out and restore the button but does not for some reason. I will have to fix that in liblxi.

lundmar, 2022-02-03:
I have released a new version of liblxi which fixes the bug. Now you will get an error when trying to send using a TCP/RAW connection on a port which the target instrument does not support.

closed 2022-02-03 by lundmar