lxi-tools/lxi-tools #42: Rigol DM3068 not discoverable

sighenterausername, 2022-02-04:
The Rigol DM3068 is not responding with the command lxi discover, neither default or mDNS.

Explanation from EEVblog user Pedre:

I don't know if it's the bug or how the LXI tools work, but just a remark about the DM3068 and the VXI protocol.
The DM3068 can not be accessed when using device flags at e.g. 'device_write'. I had used the flag 'WaitLock' and the communication was denied. The Rigol DG4000 with old firmware also gives an error message on the screen. Without device flags everything works without problems.

https://www.eevblog.com/forum/testgear/open-source-lxi-tools-v2-0-released/

lundmar, 2022-02-04:
Thanks for reporting it. I will get back to you once I've fixed some other issues.

sighenterausername, 2022-02-05:
Many thanks, happy to try help out with my own DM3068 if it helps. The DMM is reachable with 
lxi scpi -a <ipaddress> *IDN?

Also lxi benchmark -a <IP address>  and screenshot function work.

lundmar, 2022-02-05:
I trust you have the latest firmware installed?

lundmar, 2022-02-05:
Also, I just want to make sure. You have other instruments on the exact same network that responds fine to the lxi-tools search function? If there is a router or something in between it may not allow the broadcast messages come through to the instrument.

sighenterausername, 2022-02-05:
Yes, there is a Netgear ProSafe JSG516PE 16 port switch in between. Ill try again with without the switch.

lundmar, 2022-02-05:
A switch should work fine. A router router routing from one subnet to anther may be configured to not allow the broadcast packets. If other instruments responds fine on the exact same network subnet then your network is probably fine.

lundmar, 2022-02-05:
Also, if you are running lxi-tools in a VM, you need to make sure that it uses a bridged network adapter configuration (not NAT), else the broadcast message will not go through.

sighenterausername, 2022-02-05:
Yeah, the other 5 devices work fine through the switch, just the DM3068 that doesn't discover. No, it's on a RPi, indeed, with VMware fusion on my Mac I need to select bridged otherwise NoJoy.

lundmar, 2022-02-05:
Hmm, how about your instrument firmware? Running the latest?

lundmar, 2022-02-05:
If you have access to a Windows machine... please try install this good ole windows program:

https://www.lxistandard.org/About/LXI-Discovery-Tools.aspx

... and see if it finds your DM3068. If not, then we know there is something wrong with your instrument or setup. If it finds it, then we know there is something wrong with lxi-tools.

My lxi-tools discovery implementation should mimic the same behaviour as this reference LXI discovery tool.

sighenterausername, 2022-02-05:
Nope, doesn't find it and it does find the other devices... So indeed, must be the setup.
OK, found it! I had it set to DHCP OFF and a manual IP address, I switched DHCP and auto IP back on, ran discover and now it shows up.. Sincere apologies for waisting your precious time.

lundmar, 2022-02-06:
Ha ha, no apologies needed. I could have given you better advice earlier but I needed to refresh my memory around the DM3068. Once I remembered that someone reported it working long time ago and I knew we hadn't changed the lxi-tools discover implementation then I knew something must be wrong in your setup.

I'm glad we got it resolved and that lxi-tools works as intended.

closed 2022-02-06 by lundmar