lxi-tools/lxi-tools #43: lxi-gui not starting on Ubuntu or RaspberryPi OS

sighenterausername, 2022-02-04:
When trying to start lxi-gui on current as per 4 feb 2022 Linux distributions Ubuntu or Raspberry PiOS (via Snap) the following error message returns:

(lxi-gui:2313): Gdk-ERROR **: 16:44:18.660: The program 'lxi-gui' received an X Window System error.
This probably reflects a bug in the program.
The error was 'GLXBadFBConfig'.
  (Details: serial 2141 error_code 167 request_code 152 (GLX) minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

lundmar, 2022-02-04:
Please provide details about which version Ubuntu and RPI OS you are using.

sighenterausername, 2022-02-05:
Ubuntu 21.10 64bit and Raspberry Pi OS (64 bit), released 28/01/2022. Both on a Raspberry Pi 4 8GB.

lundmar, 2022-02-05:
Can you please provide the full output of lxi-gui?

I believe there should be some more preceding lines before that error.

sighenterausername, 2022-02-05:
Sorry but no, this is all I'm getting. If I try to run it as root it does not recognise the command at all.

lundmar, 2022-02-06:
Ok, I've tried making a change to the latest snap which may fix the issue.

Please try run a 'snap refresh'. On ARM64 you should be running snap #1483. See 'snap info lxi-tools'.

sighenterausername, 2022-02-06:
[image]

sighenterausername, 2022-02-06:
Also on Raspberry Pi OS still the same error. I installed an Ubuntu AMD64 VMware VM on my Mac, tried the same and there it _does_ work. RPi is hooked directly to the switch, same for the MacBookPro. Looks like its related to ARM64, or my RPi setup. Perhaps 64 bit not stable enough. Will try with 32 bit versions later today.

lundmar, 2022-02-06:
Can you try run it like so:
```
$ GDK_SYNCHRONIZE=1 lxi-gui
```

sighenterausername, 2022-02-06:
same error

lundmar, 2022-02-06:
Exact same output?

Btw. are you running it as normal user or root user?

sighenterausername, 2022-02-06:
exact same output. normal user. When running sudo lxi-gui also same error. When running SUDO GDK_SYNCHRONIZE=1 lxi-gui output is different. Then I get "Authorization required, but no authorization protocol specified". 

Cannot open display:  :0

lundmar, 2022-02-06:
I just wanted to make sure you are not running as root. That would be no good.

Try the following:
```
$ GDK_BACKEND=x11 lxi-gui
$ GDK_BACKEND=wayland lxi-gui
```
Also, try run using lxi-tools.lxi-gui instead of lxi-gui

sighenterausername, 2022-02-06:
All not working, all with the same error. either wayland or x11, with or without lxi-tools.lxi-gui  
Again, will try 32 bet versions later, perhaps 64 bit is just not mature enough. I'm happy enough it works on AMD64 in a VM

lundmar, 2022-02-10:
Please try the latest snap edge version for arm64, it may have solved the issue:
```
$ snap refresh lxi-tools --edge
```

sighenterausername, 2022-02-14:
[image]
as user

sighenterausername, 2022-02-14:
[image]
As root

lundmar, 2022-02-14:
Great. You can ignore the remaining error messages - they are not critical. Also, you should be able to use the snap stable channel now.