lxi-tools/lxi-tools #45: Req: Expanding the section '3.3 Installation from source'

hooperbloob, 2022-02-08:
I have some instruments that are throwing errors and not providing screenshots and I'd like to debug the application as it runs and edit the commands sent to them.   Could someone expand on the build instructions beyond what is already shown in the readme? 

Perhaps a listing of the commands sent as a machine is loaded with the dependencies (appropriate versions, etc), how/which debugger is launched, etc.

Trying to install the dependencies on Ubuntu:

```
br@black:~$ sudo apt install libgtk

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libgtk
```

```
br@black:~$ sudo apt install lua

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package lua is a virtual package provided by:
  lua5.3:i386 5.3.3-1.1ubuntu2
  lua5.2:i386 5.2.4-1.1build3
  lua5.1:i386 5.1.5-8.1build4
  lua50 5.0.3-8build1
  lua5.3 5.3.3-1.1ubuntu2
  lua5.2 5.2.4-1.1build3
  lua5.1 5.1.5-8.1build4
You should explicitly select one to install.
E: Package 'lua' has no installation candidate
```

```
br@black:~$ sudo apt install libreadline
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libreadline
```

The installation meson section looks straightforward once the above is sorted  out.  Then which ide/debugger to use after that?

lundmar, 2022-02-08:
Oh, I'm sorry. The from source instructions are very generic because it is simply impossible make one description that fits all systems - libraries have slightly different names on different systems. I really don't want to put specific instructions for how to build on Ubuntu, Fedora, MacOS, Arch Linux, BSD, etc.. In short, if you build stuff from source you best have experience with meson as mentioned in the foot note of the readme.

I can help you setup your debug environment on Ubuntu. However, to begin with, I don't think you need to build lxi-gui to debug screenshot issues. The screenshot mechanism is common to both lxi and lxi-gui. So you could simply build only lxi which does not require a lot of the dependencies.

Also, try start lxi-gui in your console and see if there are any error messages printed to your console when the screenshot fails. Likely it fails to find an appropriate plugin or maybe it loads a plugin which does not support your instrument.

lundmar, 2022-02-09:
Did you have a look at the cmd line to look for any error messages?

closed 2022-02-21 by lundmar