lxi-tools/lxi-tools #46: Requirement 'lua-5.3' doesn't work across Linux distributions

robert-scheck, 2022-02-20:
The requirement `lua-5.3` in [src/meson.build](https://github.com/lxi-tools/lxi-tools/blob/master/src/meson.build) doesn't work across Linux distributions. For Fedora and RHEL, this needs to be changed to `lua` to get it building. On the same hand this raises the question for supported Lua versions. With lxi-tools 1.x, Lua >= 5.1 was supported (which satisfies e.g. RHEL 7 and later). However modern Linux distributions are shipping only Lua 5.4 (e.g. Fedora >= 35, RHEL 9), where `lua-5.3` would be not achievable.

lundmar, 2022-02-20:
Fixed in 8319ca0f9af088acd4f7a784a37e31c99aeb47bd

robert-scheck, 2022-02-20:
Confirming, commit 8319ca0f9af088acd4f7a784a37e31c99aeb47bd builds on all active Fedora and RHEL releases on all architectures.

closed 2022-02-20 by lundmar