lxi-tools/lxi-tools #48: Does not compile - missing dependencies and invalid command parameters

SheetLightning, 2022-04-23:
I had lxi-tools running on my old PC and it worked very well with my Rigol 1054Z oscilloscope so thank you for making this available. Unfortunately, I have run into several problems trying to compile this latest version from sources on my new PC. I am running Linux Mint 20.3. The first problem is dependencies.

I was able to install liblxi1 and meson.

The packages readline-common, bash-completion, libgtk-3-0, libgtksourceview-3.0-common were already installed.

However I could not find libadwaita.

There was no package called 'lua' but there was one called lua5.3. Nevertheless after installing that I still get:

```
The Meson build system
Version: 0.53.2
Source dir: /home/johnc/src/lxi-tools
Build dir: /home/johnc/src/lxi-tools/build
Build type: native build
Project name: lxi-tools
Project version: 2.1
C compiler for the host machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Configuring config.h using configuration
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Found CMake: /usr/bin/cmake (3.16.3)
Run-time dependency lua-5.4 found: NO (tried pkgconfig and cmake)
Run-time dependency lua-5.3 found: NO (tried pkgconfig and cmake)
Run-time dependency lua-5.2 found: NO (tried pkgconfig and cmake)
Run-time dependency lua-5.1 found: NO (tried pkgconfig and cmake)
Run-time dependency lua found: NO (tried pkgconfig and cmake)

src/meson.build:50:2: ERROR: Problem encountered: Lua could not be found!

A full log can be found at /home/johnc/src/lxi-tools/build/meson-logs/meson-log.txt
```
There is no global lua-dev package although there are serveral -dev packages for various components, however since I don't know which one to install, I am unable to proceed.

I also rather prematurely tried the next step which is:

meson compile -C build

This gave me:

```
$ meson compile -C build
usage: meson [-h]
             {setup,configure,dist,install,introspect,init,test,wrap,subprojects,help,rewrite}
             ...
meson: error: unrecognized arguments: -C build
```
So there seems to be no parameter option -C. The installed version of meson seems to be 0.53.2 which appears to be the latest available in the Mint repository. I am curious why traditional tools such as cmake and make have not been used? I have never heard of meson but a quick google search seems to indicate that apart from a subatomic particle, it is yet another build system. Sigh :-(. I must admit to being a bit reluctant to install yet more bloat on my system.

Obviously the lua and libadwaita dependencies need to be solved first so I would like to ask which packages I need to install? I do not recall having any such issues when I previously installed lxi-tools it on my old PC, although that was some time ago now.

Incidentally, I found lxi-tools in the Mint repository and I installed this first as it is the simplest option, however, it seems to be missing the lxi-gui tool for some reason? Even an older outdated version of the GUI would have been useful as it did previously work just fine and did what I needed.

lundmar, 2022-04-23:
The new lxi-gui application uses very cutting edge versions of various dependencies such as gtk4, gtksourceview5, libadwaita1, etc. For this reason you will have a hard time trying to compile lxi-gui on any distro except maybe the newly released Ubuntu 22.04.

For this reason I recommend you simply go ahead and install lxi-tools via snap - it is the easiest way.

I understand your frustration with yet another build system but I'm afraid we can't dwell on old build systems such as autotools and cmake forever when new and much better build systems such as meson is quickly becoming popular for modern application development. I'm saying this as someone who have been using autotools for decades and suffered the pain of maintaining and distributing software via autotools.

lundmar, 2022-04-23:
Btw. for lua dependency you want to install liblua-dev

SheetLightning, 2022-04-23:
Well, one step further so thank you for that. Installing the liblua5.3-dev package sorts out the lua dependency.

I already have liblxi1 installed and tried installing liblxi-dev but meson still complains it can't find liblxi:

```
$ meson compile
The Meson build system
Version: 0.53.2
Source dir: /home/johnc/src/lxi-tools
Build dir: /home/johnc/src/lxi-tools/compile
Build type: native build
Project name: lxi-tools
Project version: 2.1
C compiler for the host machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Configuring config.h using configuration
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Found CMake: /usr/bin/cmake (3.16.3)
Run-time dependency lua-5.4 found: NO (tried pkgconfig and cmake)
Run-time dependency lua-5.3 found: YES 5.3.3
Library readline found: YES
Run-time dependency liblxi found: NO (tried pkgconfig and cmake)

src/meson.build:55:0: ERROR: Dependency "liblxi" not found, tried pkgconfig and cmake

A full log can be found at /home/johnc/src/lxi-tools/compile/meson-logs/meson-log.txt
```
I also tried running ldconfig just in case.
Incidentally i am running 'meson compile' rather than 'meson compile -C build.
Is there a meson equivalent of make clean?

I am in two minds about snap. I am told its not a good idea to mix different methods of installing packages on a system. Besides, snap seems to have huge packages and the last thing I need is extra bloat. I have tried the odd appimage in the past and that has not been without its problems. Having said that' snap is available for my distro.

SheetLightning, 2022-04-23:
UPDATE - I discovered liblxi on your GitHub so removed the one installed from the repository. Instead I compiled and installed your version. Running meson in the lxi-tools source directory now found liblxi and I was then able to complete building and install lxi-tools.

lxi runs, but I still have no lxi-gui?
Has this been renamed, or is there a separate application download?

lundmar, 2022-04-23:
Yes, lxi-tools require a more recent version of liblxi than the one Mint offers.

To build lxi-gui you need to enable it during build a la "meson build -Dgui=true"

Again, I highly recommend using the snap. You are in for a world of hurt in the upcoming compile steps.

Snap does not break your system packages - they coexist.

SheetLightning, 2022-04-23:
Hmm, it seems so. I discovered in the meantime that lxi-gui had to be enabled in meson_options.txt. I can now understand why it is disabled by default!

You already pointed out the gtk4 dependency, but it also requires a later version of meson which is also unavailable to this distro.

It looks like I will have to give up at this point. I appreciate your replying to me but I guess I will not be able to use the GUI version of the application. It is a shame that it is no longer accessible to me as it worked very well previously. I might still have the old sources on the other computer so that may be a way forward. I am sorry but I do not intend to use the snap image. I simply cannot afford to be that extravagant with SSD space. Although I can understand why you are recommending it, I think the advantages are outweighed by the disadvantage of huge bloat.

Thank you for your help though.

lundmar, 2022-04-23:
You can always download and compile older versions of lxi-tools until your Mint distribution catches up with more up-to-date dependencies so you can move forward with newer lxi-tools. All releases of liblxi and lxi-tools are available for download.

closed 2022-04-23 by lundmar