lxi-tools/lxi-tools #53: Does it support windows?

OPISGIT, 2022-07-27:
I need a scpi tool to run at Windows 10.
Is this supporting?

lundmar, 2022-07-27:
You can install the WSL2 (Windows Subsystem for Linux) and e.g. Ubuntu from the Windows app store. This will give you an Ubuntu command-line via which you can install and run lxi-tools using the command 'apt-get install lxi-tools'. From there you should be able to run the lxi command to discover instruments and send SCPI commands and scripts etc.

It should be fairly easy to get WSL2/Ubuntu up and running but I haven't tried myself as I don't use Windows.

You may even be able to run the graphical application lxi-gui but I don't think it is available on Ubuntu yet.

closed 2022-07-28 by lundmar

thestumbler, 2023-10-07:
I gave this a very quick try a few days ago. My first attempt failed, but that was because I was using WSL1 vs WSL2 (an Ubuntu 22.04 LTS). I will give WSL2 a try soon. 

I'm curious what is it about WSL1 that keeps the GUI from installing and working? I can run GUI tools from WSL1 using a X-client on Windows. Is it the USB support? 

When I tried to install with SNAP as some folks online suggest, it gives me the error "WSL1 not supported". Installing via APT worked but only the command line tools. It's possible the tools only installed but didn't work, because I wasn't connected to my oscilloscope at the time to test further.

lundmar, 2023-10-07:
@thestumbler lxi-gui is only included in Ubuntu 23.04 and onward. To install the latest version you can use snap but snap is only available in WSL2 (because snap requires systemd etc.).