lxi-tools/lxi-tools #59: Desired downstream (RPM) package name for lxi-gui?

robert-scheck, 2022-10-01:
So far lxi-tools (the command-line tool `/usr/bin/lxi`) are packaged as lxi-tools RPM for Fedora and EPEL. Given the software requirements for the `/usr/bin/lxi-gui` tool in lxi-tools v2.1 are meanwhile satisfied by latest Fedora releases, I wonder what the desired downstream (RPM) package name for lxi-gui would be - e.g. lxi-tools-gui or simply just lxi-gui? I would like to avoid putting `/usr/bin/lxi-gui` into the existing lxi-tools RPM, given it depends on much more (graphical) libraries that maybe not everybody would like to install when just using the command-line tool `/usr/bin/lxi`.

And: Are you aware about any distribution/downstream having `/usr/bin/lxi-gui` already packaged?

lundmar, 2022-10-01:
The lxi-tools project includes both the lxi and lxi-gui tool and for that reason I think it best that any lxi-tools package includes both too when possible.

I don't think going down the path of creating a separate lxi-tools-gui or lxi-gui package is a good idea, not even if it saves the command-line user from installing the GTK4 libraries because most users will likely already have GTK4 installed one way or the other because many other applications depend on them.

I think AUR have lxi-gui packaged in their lxi-tools-git package.

FYI - in the next version of lxi-tools I'm going to enable build of the lxi-gui application by default because most distribution now ship all the required GTK4 etc. dependencies.

robert-scheck, 2022-10-02:
> I don't think going down the path of creating a separate lxi-tools-gui or lxi-gui package is a good idea, not even if it saves the command-line user from installing the GTK4 libraries because most [...]

Okay…then I'm putting `/usr/bin/lxi` and `/usr/bin/lxi-gui` into the same RPM package.

> FYI - in the next version of lxi-tools I'm going to enable build of the lxi-gui application by default because most distributions now ship all the required GTK4 etc. dependencies.

For the record: Fedora 36, released on 2022-05-10, is the very first Fedora version that matches the quite high requirements. 

May I assume that it's still possible to disable `/usr/bin/lxi-gui` during build-time in future versions to still cover older Fedora and EPEL releases with the command-line application only?

lundmar, 2022-10-02:
> For the record: Fedora 36, released on 2022-05-10, is the very first Fedora version that matches the quite high requirements.

Yes, the requirements are quite high as lxi-gui uses all the latest bells ans whistles that GTK4 and libadwaita offers but at least now the various distributions are catching up with GTK4 etc. so it should be easy for most to start including lxi-gui and that is why I'm going to enable it by default.

> May I assume that it's still possible to disable `/usr/bin/lxi-gui` during build-time in future versions to still cover older Fedora and EPEL releases with the command-line application only?

Yes, for those environments that can't fulfill the build requirements there is a meson build option to disable installation of lxi-gui. It is described [here](https://github.com/lxi-tools/lxi-tools/blob/master/meson_options.txt). In short, you just add `-Dgui=false` to the meson setup line to disable build of the lxi-gui application.

robert-scheck, 2022-10-02:
Thank you for the clarification.

For Fedora 36 and the upcoming Fedora 37, [FEDORA-2022-d3250d3d9c](https://bodhi.fedoraproject.org/updates/FEDORA-2022-d3250d3d9c) and [FEDORA-2022-492ab6fc04](https://bodhi.fedoraproject.org/updates/FEDORA-2022-492ab6fc04) provide now lxi-tools with `/usr/bin/lxi-gui`; feedback is appreciated.

closed 2022-10-02 by robert-scheck

lundmar, 2022-10-02:
Great - thanks for including the lxi-gui application.

robert-scheck, 2022-10-15:
Sadly commit 8a6abce234809ff4d83e11f6a07016d56b9e8388 reduces this now to Fedora 37 only, as Fedora 36 ships 1.1.4.

lundmar, 2022-10-15:
I know but I really want to keep moving lxi-tools forward according to Gnome HIG guidelines and libadwaita so no v2.2 gui app in F36 is fine with me.