lxi-tools/lxi-tools #61: Undefined reference to `luaL_setfuncs'

robert-scheck, 2022-12-15:
Building lxi-tools v2.4 using Lua 5.1 fails like this:

```
+ /usr/bin/meson compile -C x86_64-redhat-linux-gnu -j 6 --verbose
ninja: Entering directory `x86_64-redhat-linux-gnu'
[1/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/benchmark.c.o -MF src/lxi.p/benchmark.c.o.d -o src/lxi.p/benchmark.c.o -c ../src/benchmark.c
[2/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/discover.c.o -MF src/lxi.p/discover.c.o.d -o src/lxi.p/discover.c.o -c ../src/discover.c
[3/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/run.c.o -MF src/lxi.p/run.c.o.d -o src/lxi.p/run.c.o -c ../src/run.c
[4/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/main.c.o -MF src/lxi.p/main.c.o.d -o src/lxi.p/main.c.o -c ../src/main.c
[5/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/lxilua.c.o -MF src/lxi.p/lxilua.c.o.d -o src/lxi.p/lxilua.c.o -c ../src/lxilua.c
../src/lxilua.c: In function 'lua_register_lxi':
../src/lxilua.c:418:5: warning: implicit declaration of function 'luaL_setfuncs' [-Wimplicit-function-declaration]
     luaL_setfuncs(L, lxi_lib, 0);
     ^
[6/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/misc.c.o -MF src/lxi.p/misc.c.o.d -o src/lxi.p/misc.c.o -c ../src/misc.c
[7/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/scpi.c.o -MF src/lxi.p/scpi.c.o.d -o src/lxi.p/scpi.c.o -c ../src/scpi.c
[8/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_keysight-dmm.c.o -MF src/lxi.p/plugins_screenshot_keysight-dmm.c.o.d -o src/lxi.p/plugins_screenshot_keysight-dmm.c.o -c ../src/plugins/screenshot_keysight-dmm.c
[9/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/options.c.o -MF src/lxi.p/options.c.o.d -o src/lxi.p/options.c.o -c ../src/options.c
[10/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_siglent-sdg.c.o -MF src/lxi.p/plugins_screenshot_siglent-sdg.c.o.d -o src/lxi.p/plugins_screenshot_siglent-sdg.c.o -c ../src/plugins/screenshot_siglent-sdg.c
[11/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/screenshot.c.o -MF src/lxi.p/screenshot.c.o.d -o src/lxi.p/screenshot.c.o -c ../src/screenshot.c
[12/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_rigol-dl3000.c.o -MF src/lxi.p/plugins_screenshot_rigol-dl3000.c.o.d -o src/lxi.p/plugins_screenshot_rigol-dl3000.c.o -c ../src/plugins/screenshot_rigol-dl3000.c
[13/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_keysight-ivx.c.o -MF src/lxi.p/plugins_screenshot_keysight-ivx.c.o.d -o src/lxi.p/plugins_screenshot_keysight-ivx.c.o -c ../src/plugins/screenshot_keysight-ivx.c
[14/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_lecroy.c.o -MF src/lxi.p/plugins_screenshot_lecroy.c.o.d -o src/lxi.p/plugins_screenshot_lecroy.c.o -c ../src/plugins/screenshot_lecroy.c
[15/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_siglent-sdm3000.c.o -MF src/lxi.p/plugins_screenshot_siglent-sdm3000.c.o.d -o src/lxi.p/plugins_screenshot_siglent-sdm3000.c.o -c ../src/plugins/screenshot_siglent-sdm3000.c
[16/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_rigol-dm3068.c.o -MF src/lxi.p/plugins_screenshot_rigol-dm3068.c.o.d -o src/lxi.p/plugins_screenshot_rigol-dm3068.c.o -c ../src/plugins/screenshot_rigol-dm3068.c
[17/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_rigol-1000z.c.o -MF src/lxi.p/plugins_screenshot_rigol-1000z.c.o.d -o src/lxi.p/plugins_screenshot_rigol-1000z.c.o -c ../src/plugins/screenshot_rigol-1000z.c
[18/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_rigol-2000.c.o -MF src/lxi.p/plugins_screenshot_rigol-2000.c.o.d -o src/lxi.p/plugins_screenshot_rigol-2000.c.o -c ../src/plugins/screenshot_rigol-2000.c
[19/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_rigol-dp800.c.o -MF src/lxi.p/plugins_screenshot_rigol-dp800.c.o.d -o src/lxi.p/plugins_screenshot_rigol-dp800.c.o -c ../src/plugins/screenshot_rigol-dp800.c
[20/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_siglent-sds.c.o -MF src/lxi.p/plugins_screenshot_siglent-sds.c.o.d -o src/lxi.p/plugins_screenshot_siglent-sds.c.o -c ../src/plugins/screenshot_siglent-sds.c
[21/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_screenshot_rigol-dsa.c.o -MF src/lxi.p/plugins_screenshot_rigol-dsa.c.o.d -o src/lxi.p/plugins_screenshot_rigol-dsa.c.o -c ../src/plugins/screenshot_rigol-dsa.c
[22/28] cc -Isrc/lxi.p -Isrc -I../src -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -MD -MQ src/lxi.p/plugins_s [... cut ...] s-needed -Wl,--allow-shlib-undefined -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -Wl,-z,relro -Wl,--start-group -lreadline /usr/lib64/liblxi.so -lm -ldl -llua -Wl,--end-group
src/lxi.p/lxilua.c.o: In function `lua_register_lxi':
/builddir/build/BUILD/lxi-tools-2.4/x86_64-redhat-linux-gnu/../src/lxilua.c:418: undefined reference to `luaL_setfuncs'
collect2: error: ld returned 1 exit status
```

lundmar, 2022-12-15:
Thanks for the hint. I've added the function as described to make it backwards compatible with 5.0/5.1 in 47aa53ddd8db03e2b47c0b24d0bfab838b184877

lundmar, 2022-12-15:
No problem 👍🏻  Thanks for reporting the issue.