lxi-tools/lxi-tools #63: Unable to read image data from Rigol MSO5000

Twilight-Logic, 2022-12-22:
I know its not on your supported devices list, but I am unable to read the screen image data from the MSO5000 - at least with the latest version - 2.4.

Long story short: I eventually managed to get lxi-tools 2.4 working on my machine, having previously used 1.21 CLI which was the version available from the repository. Version 1.21 captured images from both the Rigol DS1054Z and my new MSO5000. Version 2.4 captures screenshots from the DS1054Z, but not the MSO5000 and retrurns:

`Failure handling image format
`
Analysis of the data sent using Wireshark shows that in both cases the command sent by lxi-tools is the same:

`display:data? on,0,png`

I am unsure whether the part after the '?' is a command parameter or not, but both oscilloscopes appear to send data as a bitmap (BMP). The returned image header is slightly different for each scope:

```
DS1054Z -  #9001152054BM6...
MSO5000 - #9001843254BM6...

```
This is not surprising since the screen size of the MSO5000 is larger and therefore it sends more image data. It could be a simple matter of buffer size I suppose, but the manuals of neither oscilloscope indicate an option to change the format of the image data sent via SCPI. One therefore assumes that 'on,0,png' is simply ignored. Yet it is curious that the same SCPI command works in version 1.21 but not in version 2.4.

I am happy to provide any information/data required to help make image capture work again in version 2.4.

Regards.

lundmar, 2022-12-22:
Can you try build the latest lxi-tools CLI and try extract the screenshot that way?

Twilight-Logic, 2022-12-22:
Thanks. I had the same thought to run version 2.4 of lxi CLI and would certainly like to oblige, but I have run into difficulty. Firstly, the project  requires a later version of meson than 0.53.2 which Linux Mint tells me is the latest available in their repository:

`subprojects/gtk-4.6.0/meson.build:1:0: ERROR: Meson version is 0.53.2 but project requires >= 0.59`

I removed the meson package from the official Mint respository and installed it via pip but when I try to run 'meson' I get an error telling me:

`bash: /usr/bin/meson: No such file or directory`

I have ./local/bin in my $PATH and meson is present in there and I have tried sudo ldconfig but no change. Not sure why I am getting this error. I can however run it as ~/.local/bin/meson. So next, I deleted the lxi-tools directory that was cloned previously from git and cloned a fresh copy. I then went into that directory and ran ~/local/bin/meson build and got this:

```
libtiff| Subproject libtiff finished.

gtk| Dependency libtiff-4 from subproject subprojects/tiff-4.1.0 found: YES undefined
gtk| Run-time dependency libjpeg found: NO (tried pkgconfig and cmake)
gtk| Looking for a fallback subproject for the dependency libjpeg

subprojects/gtk-4.6.0/meson.build:422:0: ERROR: Neither a subproject directory nor a libjpeg-turbo.wrap file was found.

A full log can be found at /home/johnc/src/lxi-tools/build/meson-logs/meson-log.txt
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.

```

libjpeg9 and libjpeg9-dev are installed. I am not sure what I need to do next?

lundmar, 2022-12-22:
There is a lot of complications here so I simply recommend you to install snap which is supported on Linux Mint (https://snapcraft.io/docs/installing-snap-on-linux-mint). Then install latest lxi-tools via snap.

Twilight-Logic, 2022-12-23:
I am trying to avoid the SNAP package as its not supported by the majority of Linux distros including Linux Mint (although it can be enabled as you point out) and MX Linux which are the two distros I use. Most Linux distros do however support Flatpack and I have compiled lxi-tools from your lxi-tool.flatpack repository. This is how I am able to run the GUI at all. Although there exists a compiled lxi CLI binary file in the flatpack directory, I seem to be unable to run it.

lundmar, 2022-12-23:
Yes, I know there is a lot of controversy regarding snap. However, most people do not realize  that flatpak is only designed to distribute GUI applications. This means that snap is the easy choice for instant distribution of both graphical and CLI applications. I still support the traditional way of distribution via distro packages but that is a slower path so for instant and reliable distribution and testing situations like this, snap is a good choice. And don't worry, snap will not compromise your system binaries, it will coexist.

Twilight-Logic, 2022-12-23:
Hmm, I don't think I should NEED to have to install a SNAP or Flatpack to compile and run a CLI!
It turns out however, that I CAN run the CLI from the compiled Flatpack with:

`flatpak run --command=lxi io.github.lxi-tools`

Adding --version confirms that this is running lxi v2.4:

```
$ flatpak run --command=lxi io.github.lxi-tools --version
lxi v2.4
```

So I next did:

```
$ flatpak run --command=lxi io.github.lxi-tools screenshot -a 10.210.210.163
Loaded rigol-1000z screenshot plugin
Saved screenshot image to screenshot_10.210.210.163_2022-12-23T10:36:40.png

```
with the result indicated. The image is saved as a PNG but the Wireshark dump seems to indicate it is being sent as a bitmap. Interesting though that the CLI tool appears to work.

lundmar, 2022-12-23:
The problem is in the build of the GUI application. The GUI application fails to handle the downloaded image data because of either missing or incorrectly loaded gdk-pixbuf plugins (image format handling plugins). This is fixed in the snap build.

Regarding the downloaded image format. The screenshot plugins should automatically set the format before downloading so whatever you set manually on the scope will be overridden. That being said, the GUI application will always save to PNG by image conversion regardless of the downloaded format.

I haven't studied the Rigol MSO5000 doc but I expect one of the existing screenshot plugins can be reused or modified to officially support the MSO5000.

lundmar, 2022-12-23:
> Hmm, I would not have thought it necessary to have to install a SNAP or Flatpack to compile a CLI!

That is of course not necessary but if you are not familiar with meson then I do not recommend going down that path.

Twilight-Logic, 2022-12-23:
My difficulty is partly down to the fact that I am new to meson, snap and flatpak and also because Linux Mint and MX Linux devs generally seem to discourage using snap. I have also read something regarding the controversy. Due to the way both snap and flatpak work, they create very large packages which does not help but I think I understand the convenience, especially when distro repository packages are several versions behind what the developer is using.

Those 'n,0,png' parameters are not documented in the DS1054Z programming manual, although it does provide a reasonably detailed explanation of the DISPLAY:DATA? SCPI command, including some discussion of the image format. By contrast, the MSO5000 programming manual has very little to say, providing only a very brief summary and is therefore less helpful.

I e-mailed Rigol tech support to find out whether the MSO5000 can be configured to send the screenshot image in PNG format. The Storage -> Save Image -> Format options do not seem to affect the format being sent over SCPI, only when saving to USB or internal storage. The default format is actually set to .png, but the default SCPI format seems to be BMP.

lundmar, 2022-12-23:
> My difficulty and hesitance is partly down to the fact that I am new to meson, snap and flatpak and also because Linux Mint and MX Linux devs generally seem to discourage using snap. I have also read something regarding the controversy surrounding snap and Canonical. Besides, due to the way both snap and flatpak work, they create very large packages which is not helpful but on the other hand I think I understand the convenience, especially when distro repository packages are several versions behind what the developer is using.

Well, either have it slow via traditional distributions paths or quick via containerized distribution tech like snap. Pick your poison :)

As a maintainer of open source software and having experience with both flatpak and snap, I can tell you that snap is much easier to use and maintain from a maintainers point of view and that is why I only maintain snap now. Also, snap is the only framework through which I can distribute updates to users instantly - literally within seconds. For comparison, flatpak has to go through human reviews because of its lesser secure distribution and runtime model. I think the only truly justified criticism of snap is their insistence on using one central application repository managed by Canonical themselves but that can also be considered a convenient feature. That is where the politics and debates become difficult but I think maybe also a bit too hysterical.

> 
> Those 'n,0,png' parameters are not documented in the DS1054Z programming manual, although it does provide a reasonably detailed explanation of the DISPLAY:DATA? SCPI command, including some discussion of the image format header (at least for BM6). By contrast, the MSO5000 programming manual has very little to say, providing only a very brief summary and is therefore less helpful.
> 
Yes, I think I vaguely recall this being undocumented.

> I e-mailed Rigol tech support to find out whether the MSO5000 can be configured to send the screenshot image in PNG format. The Storage -> Save Image -> Format options do not seem to affect the format being sent over SCPI, only when saving to USB or internal storage. The default format is actually set to .png, but the default SCPI format seems to be BMP. If I hear back from them with anything useful, I will pass that along.

You got the latest version working via CLI and I see that it used the rigol-1000z screenshot plugin. This means that the MSO5000 supports the exact same SCPI download command and that it actually downloaded and saved the image as PNG. The CLI tool does not do any image conversion - it simply dumps the downloaded image data to file. Only the GUI application always converts to PNG.

Twilight-Logic, 2022-12-23:
Hmm, I think we are being deceived...

```
$ identify screenshot_10.210.210.163_2022-12-23T11:13:06.png
screenshot_10.210.210.163_2022-12-23T11:13:06.png BMP3 1024x600 1024x600+0+0 8-bit sRGB 1.75786MiB 0.000u 0:00.010
identify-im6.q16: length and filesize do not match `screenshot_10.210.210.163_2022-12-23T11:13:06.png' @ warning/bmp.c/ReadBMPImage/839.

```

The file extension is .png, but the content is BMP. Apparently Xviewer doesn't care and shows it anyway...
Looking at your source in https://github.com/lxi-tools/lxi-tools/blob/fbffca6ffb56db0de26188c32a49756b1352066a/src/screenshot.c the screenshot_file_dump function just grabs the incoming data and saves it to a file with a .png extension.

So yes, evidently the CLI tool is not doing any conversion but my Linux distro is deceiving me....
The program is saving BMP format data into a file with a .png extension and Xviewer opens and shows it regardless without warning when the file is double-clicked.

Xviewer shows the properties incorrectly, but GIMP tells the real story.

[image]
[image]

lundmar, 2022-12-23:
Right, so a fox in sheep's clothing :)

It may be that the MSO5000 acts like their MSO2000 which does not support changing the screenshot format but simply assumes BMP like the rigol-2000 plugin does, ignoring any extra options.

Try fire the following command:
```
$ lxi screenshot -p rigol-2000 -a <IP>
```

If that works and you get a .bmp file then we can simply modify the rigol-2000 plugin to also include support for MSO5000 (simply append ' MSO5...' to the .regex line in the plugin definition so that it will win the auto detection).

Either way, let me know if you get a response from Rigol.

Twilight-Logic, 2022-12-23:
Yes that works!

$ lxi screenshot -p rigol-2000 -a <IP>

This suggested command captures using the rigol-2000 plugin and correctly saves the data to a file with a .bmp extension.

That being the case, I figured out how to make the change locally and recompile the Flatpak. Firstly I edited the regex line the source file, ./lxi-tools/src/plugins/screenshot_rigol-2000.c, so that it now reads:

`    .regex = "RIGOL TECHNOLOGIES Rigol Technologies DS2... MSO2... MSO5...",
`
It was then a matter of going into the  lxi-tools.flatpack directory and changing the lxi-gui section of the .yaml file to:

```
- name: lxi-gui
    sources:
      - type: dir
        path: /home/username/src/lxi-tools/
    buildsystem: meson
    config-opts:
      - "-Dgui=true"
```
The path is wherever the lxi-tools was git cloned to. Next I ran:

`$ flatpak-builder --user --install --force-clean build io.github.lxi-tools.yaml
`
which cleared the previous build and then re-built the package.

After that I initially ran the CLI:

```
$ ./lxi.sh screenshot -a 10.210.210.163
Loaded rigol-2000 screenshot plugin
Saved screenshot image to screenshot_10.210.210.163_2022-12-23T14:23:13.bmp
```
confirming that it now automatically selects the rigol-2000 plugin. Finally I ran the GUI and attempted to grab a scrteenshot:

`flatpak run io.github.lxi-tools`

I am glad to report that that your suggested modification does allow the GUI to captures screenshots from the MSO5000!

lundmar, 2022-12-23:
Great, feel free to make a github pull request with the .regex addition to the rigol-2000 plugin so we have it working.

Later on, I will rename the plugin to rigol-2k3k or similar.

Twilight-Logic, 2022-12-23:
Interestingly when I save the screenshot in the GUI this does save it as a PNG which is great as the file size is much smaller.

I have now forked the repository and created a pull request.

lundmar, 2022-12-23:
The GUI application now works because when the rigol-2000 plugin is picked it delivers BMP image data and that works well with gdk-pixbuf buildint BMP support. However, when it picked the rigol-1000z plugin it delivers PNG image data and that fails for the flatpak build because the gdk-pixbuf PNG plugin is not installed properly so it fails to handle the PNG image correctly and hence the error message. It is a bit of annoyance and I've fixed in before for the snap build so it installs the gdk-pixbuf plugins correctly to properly handle any format image data. And yes, the GUI always outputs PNG, simply because it was possible.

I've merged your PR and added you to our list of contributors. Thanks for making MSO5000 work with lxi-tools.

That reminds me, you should add your specific instrument to our list of supported test instruments in our README.md

Twilight-Logic, 2022-12-23:
When I went to add it I noticed that MSO5074 is already your list in README.md. I was looking at the plugin list. The instrument model list is further down. Doh! Your comment might explain why it might have been tested and working before. Or, perhaps only the flatpak build was affected? 

Thank you for merging the PR. 

Th model is covered by two entries:

```
Rigol Technologies MSO5074 | [discover+scpi+screenshot]

Tektronix 5 Series MSO | [discover+scpi]
```
Perhaps this reflects that only the 5074 has been tested in the series?

lundmar, 2022-12-23:
That is correct, only the flatpak build is broken. The only reason that your local flatpak build works now is because it now activates a plugin that delivers BMP data which basically bypasses the gdk-pixbuf plugin problem in lxi-gui. The snap would have worked all along.

Umm, the two listings are two different brands? Tektronix vs. Rigol...

Twilight-Logic, 2022-12-23:
My bad! Just zoomed in on the numbers I think"! Thanks for the confirmation re: flatpak.

lundmar, 2022-12-23:
No worries. To fix the general gdk-pixbuf plugin loading problem one would have to transfer some of the same build steps from the snap to the flatpak.

Either way, at least now the rigol-2000 plugin is loaded which I think is principally the better plugin to handle MSO5000. The rigol-1000z plugin is only working by chance because the scope ignores the png option stuff on the SCPI command.

lundmar, 2022-12-24:
Closing issue. Feel free to post your response from Rigol here.. if you get one ;)

closed 2022-12-24 by lundmar