lxi-tools/lxi-tools #67: report Textronix MSO56

JanSchmidt3, 2023-02-13:
- scpi works

root@770d066c1f55:/# lxi scpi --address 10.85.100.46 "*IDN?"
TEKTRONIX,MSO56,C020511,CF:91.1CT FV:1.44.3.433

- screenshot does not work

root@770d066c1f55:/# lxi screenshot --address 10.85.100.46
Loaded tektronix-2000 screenshot plugin
Error: Read error (timeout)
Error: Failed to receive message

JanSchmidt3, 2023-02-14:
this works with some modifications.
- It needs to use the double quote "
- after READFILE a question mark is required

lundmar, 2023-02-14:
Great thanks for the input.

I've created two new plugins to support your scopes. The changes are on the following branch https://github.com/lxi-tools/lxi-tools/tree/new_plugins

Would be great if you could test this branch.

JanSchmidt3, 2023-02-14:
thank you,

can you give me some more detailed compilation instruction? Prefered is a Dockerfile.
I have also access to a fresh real machine "Ubuntu 20.04.5 LTS".

(liblxi seems to be not available via apt-get)

lundmar, 2023-02-14:
On ubuntu 20.x I think something like this should suffice:
```
sudo apt install build-essential liblua5.2-dev liblxi-dev libreadline-dev meson
git clone https://github.com/lxi-tools/lxi-tools.git
cd lxi-tools
git checkout new_plugins
meson build --prefix $HOME/opt/lxi-tools -Dgui=false
ninja -C build install
```

lundmar, 2023-02-14:
Right, Ubuntu 20.x is too old so the liblxi-dev package does not include the required pkg-config stuff to be detected by meson. I suggest you try with Ubuntu 22.04 or later.

JanSchmidt3, 2023-02-16:
ok, I have a virtualbox online now.
I could compile. 
- The R&S stuff works fine.
- With Tektronix there is some trouble to read the file. Often the transfer fails with timeout. I need to test what's happening.

BTW: I tried to compile with gui and it failed.

sudo apt-get install libxkbcommon-dev libcairo2-dev libwayland-dev libxrandr-dev libxi-dev libxcursor-dev libxdamage-dev libxinerama-dev
meson build --prefix /usr/local

subprojects/gtk-4.6.0/gdk/wayland/meson.build:84:2: ERROR: Sandbox violation: Tried to grab file xdg-activation-v1.xml outside currend (sub)project.

lundmar, 2023-02-16:
>     * The R&S stuff works fine.

Great. The only downside of the current plugin implementation is that it requires an SD card to be installed for storing the temporary image file. Do you know if the RTH device has any internal memory that we could use instead?

> 
>     * With Tektronix there is some trouble to read the file. Often the transfer fails with timeout. I need to test what's happening.

If the instrument is slow it may be that the image file transfer is stopped by the timeout. Though, I see that the default timeout is 10 seconds so that would only happen if the instrument is VERY slow.

Likely the problem is something else.

> BTW: I tried to compile with gui and it failed.

I don't recommend trying to compile the UI as it is a bit more involved. It requires GKT4 and various other dependencies to be installed. If you want to know which you can take a look at src/meson.build

If you can't read the downloaded file maybe we need to strip a header from it like we do in the RTH plugin.

JanSchmidt3, 2023-02-16:
> > ```
> > * The R&S stuff works fine.
> > ```
> 
> Great. The only downside of the current plugin implementation is that it requires an SD card to be installed for storing the temporary image file. Do you know if the RTH device has any internal memory that we could use instead?
> 

I think, it's like Android. The file browser shows /media/SD and /media/USB. They name it "internal SD card". In the root folder /media/SD is the linux kernel, the device tree file and other stuff. And there is no SD card slot.
BTW: the screenshot button saves to /media/SD/Rohde-Schwarz/RTH/Screenshots

> If the instrument is slow it may be that the image file transfer is stopped by the timeout. Though, I see that the default timeout is 10 seconds so that would only happen if the instrument is VERY [...]

I think so. 
- the screenshot will be saved. I can access the file with the internal file browser.
- readfile usually fails. Sometimes it responds if before 'readfile?'  'readfile' will be send.
- sometimes the scope hangs, so that IDN? no longer gives an answer
- it seems single quote or double qoute is no difference
- the response time from the scope is <1s, if it will answer. Otherwise it ignores the command (then a timeout occures) or it does not send any data.

Meanwhile I tested other devices:
- ITECH,IT6018C power supply responds via scpi
- Goodwill PSW Series power supply does not work (it has open port 5030, if I try to connect with '-r -p 5030 scpi', the device hangs (lxi reports timeout), without '-r' the error is "unable to connect"

lundmar, 2023-02-16:
Good, then I'll consider the R&S RTH plugin working.

Regarding the Tektronix plugin maybe you could try increase the timeout value by x2 or similar to to make sure it is not a timeout issue. Maybe it is a "c:\" vs "C:\" issue. Maybe we need some spaces ";*OPC" vs "; *OPC". etc.

lundmar, 2023-02-16:
Also, maybe check the return value of lxi_send(). It returns how many bytes of the string has been sent. It may not send them all in case the target has very little receive buffer which is quite unusual.

JanSchmidt3, 2023-02-16:
[mso-5.pcapng.gz](https://github.com/lxi-tools/lxi-tools/files/10756842/mso-5.pcapng.gz)

Attached is a pcap file. Included two commands:
- screenshot, download, delete with interactive console
- screenshot with the plugin
Maybe you see the difference. What I found is, that in interactive mode
readfile? does no work, if readfile 'filename' was not sent before. But with the changed plugin source it still does not work.

lundmar, 2023-02-16:
It looks like we maybe do not receive all the data we need to form the PNG image. We may need more read call.

Also, according to the doc READFILE should not include ?. This is a bit outside of normal SCPI spec.

I've updated the new_plugins so it prints the length of received PNG image data. Can you test and see if the length is anything close to the size of the png files in the device.

JanSchmidt3, 2023-02-16:
diff --git a/src/plugins/screenshot_tektronix-mso-5.c b/src/plugins/screenshot_tektronix-mso-5.c
this works for me. I repeated it 3 times without issues.

index f654801..492db35 100644
--- a/src/plugins/screenshot_tektronix-mso-5.c
+++ b/src/plugins/screenshot_tektronix-mso-5.c
@@ -59,8 +59,12 @@ int tektronix_screenshot_mso_5(char *address, char *id, int timeout)
     // Send SCPI commands to grab PNG image
     command = "SAVE:IMAGE 'c:/lxi-tools-screenshot.png';*OPC";
     lxi_send(device, command, strlen(command), timeout);
+    sleep(1);
     command = "FILESYSTEM:READFILE 'c:/lxi-tools-screenshot.png'";
     lxi_send(device, command, strlen(command), timeout);
+    sleep(1);
+    command = "FILESYSTEM:READFILE?";
+    lxi_send(device, command, strlen(command), timeout);
     length = lxi_receive(device, response, IMAGE_SIZE_MAX, timeout);
     printf("length = %d\n", length);
     if (length < 0)

lundmar, 2023-02-16:
Hmm, maybe the command parser of the instrument does not support multiple commands in one string and so the *OPC (operation complete command) has no effect and you need the sleep.

Perhaps we can get rid of the sleep() doing something like this instead:
```
     ...
     command = "SAVE:IMAGE 'c:/lxi-tools-screenshot.png'";
     lxi_send(device, command, strlen(command), timeout);
     command = "*OPC"
     lxi_send(device, command, strlen(command), timeout);
     command = "FILESYSTEM:READFILE 'c:/lxi-tools-screenshot.png'";
     lxi_send(device, command, strlen(command), timeout);
     command = "*OPC"
     lxi_send(device, command, strlen(command), timeout);
     command = "FILESYSTEM:READFILE?";
     lxi_send(device, command, strlen(command), timeout);
     length = lxi_receive(device, response, IMAGE_SIZE_MAX, timeout);
     ...
```

JanSchmidt3, 2023-02-17:
sadly this didn't work.
The whole sequence including the two additional *OPC commands took only two milliseconds.
Same behavior as without the sleeps (timeout, sometimes the scope hangs and does not respond to connects anymore) .

lundmar, 2023-02-17:
In the documentation there is a section "Synchronization Methods" which describes how to make the *OPC command work. Apparently it involves enabling some bits in status registers.

```
/* Enable the status registers */
DESE 1
*ESE 1
*SRE 32

/* Wait until the acquisition is complete before taking the measurement*/
*OPC
```

It also describes an alternative using the *WAI command. This seems to require no setup of status registers.
```
/* Wait until the acquisition is complete before taking the measurement */
*WAI
```

I think we should try the same you just tested but simply replace *OPC with *WAI.

JanSchmidt3, 2023-02-17:
*wai didn't help. I fount another solution.

The scope sends 0 after ca. 120ms.
Additional some kind of timeout to exit the loop is required. A loop counter is not sufficient, so I suggest something like a timer.

```
diff --git a/src/plugins/screenshot_tektronix-mso-5.c b/src/plugins/screenshot_tektronix-mso-5.c
index f654801..9da8c12 100644
--- a/src/plugins/screenshot_tektronix-mso-5.c
+++ b/src/plugins/screenshot_tektronix-mso-5.c
@@ -59,10 +59,14 @@ int tektronix_screenshot_mso_5(char *address, char *id, int timeout)
     // Send SCPI commands to grab PNG image
     command = "SAVE:IMAGE 'c:/lxi-tools-screenshot.png';*OPC";
     lxi_send(device, command, strlen(command), timeout);
+    command = "BUSY?";
+    do {
+        lxi_send(device, command, strlen(command), timeout);
+        length = lxi_receive(device, response, IMAGE_SIZE_MAX, timeout);
+    } while ((length > 0) && (response[0] == '1'));
     command = "FILESYSTEM:READFILE 'c:/lxi-tools-screenshot.png'";
     lxi_send(device, command, strlen(command), timeout);
     length = lxi_receive(device, response, IMAGE_SIZE_MAX, timeout);
-    printf("length = %d\n", length);
     if (length < 0)
     {
         error_printf("Failed to receive message\n");
```

lundmar, 2023-02-17:
> *wai didn't help. I fount another solution.

Could you please try enable the status registers as I described and use the *OPC again. According to doc that should make the *OPC readable only when all events are done, meaning the *OPC call should only return once the operation is complete. 

One last try :) If that does not work I'm okay with using your polling method but perhaps with an upper loop limit so we don't risk ending up looping forever. 

> Before I tested with interactive mode. The readfile command does not call "lxi_receive", because no ? is given in the command.

Right, in interactive mode ? means it will wait for a response via lxi_receive(). This is adheres to the SCPI specs request/response model. The fact that Tektronics READFILE does not make it a proper SCPI query request by by using a ? seems to me to be a violation of the SCPI spec.

JanSchmidt3, 2023-02-17:
ok, here we go...

after system start the registers are: *sre 0; *ese 0; dese 255
changing them as you mentioned does not work. But: if the *wai is sent as a separate command, then the scope answers correct. And also *opc works - but also as extra command.

```
+    // enable Service Request Method
+//    command = "dese 1;*ese 1; *sre 32";
+//    lxi_send(device, command, strlen(command), timeout);
     // Send SCPI commands to grab PNG image
-    command = "SAVE:IMAGE 'c:/lxi-tools-screenshot.png';*OPC";
+    command = "SAVE:IMAGE 'c:/lxi-tools-screenshot.png'";
     lxi_send(device, command, strlen(command), timeout);
+    // wait until ready
+//    command = "*OPC";
+    command = "*WAI";
+    lxi_send(device, command, strlen(command), timeout);
     command = "FILESYSTEM:READFILE 'c:/lxi-tools-screenshot.png'";
```
Extra advantage: both variants are faster than polling with busy?
I think, the *wai method is the best.

lundmar, 2023-02-17:
Great. Yes, this is exactly why I wanted the *OPC or *WAI to work. They are the most optimal way. The check for BUSY adds an overhead so it will always be slower. The *wai or *opc commands sent separately via lxi_send() will wait for the command completion event to happen and then the lxi_send() will return. This is part of the VXI11 mechanism.

Lets go with the *WAI command then. I've updated the branch with the code that should work now.

JanSchmidt3, 2023-02-17:
I've updated. It works.

There is still the unneeded 
printf("length = %d\n", length);

lundmar, 2023-02-20:
Ok 👍🏻 

I've updated the list of supported instruments in 776a20ac0eda361645f31cc5ad808cf3d3f51a11 according to your findings. Let me know if I got something wrong in the listed details.

Apparently your GW PSU only allows one tcp connection and with a 1 minutes timeout on error that quickly becomes a bit tedious when errors occur. Most instruments allow multiple connections.

closed 2023-02-21 by lundmar