lxi-tools/lxi-tools #68: Vagrant config for development

hooperbloob, 2023-03-03:
With the complex configuration oft mentioned for development of the UI portion, it would be great if a 
Vagrant configuration could be provided that specifies and assembles all that. I'd like contribute and build
the MacOS version...

lundmar, 2023-03-03:
@hooperbloob You are welcome to create a Vagrant VM configuration to start working on a mac build. I can create a repository in the lxi-tools github organization for that specific purpose if you like? Much in the same way we have separate repositories for snap, flatpak, etc.

hooperbloob, 2023-03-03:
Sure, you can create the repo but I'm at a loss as to the various dependencies required. Thats why I was hoping you could create the first Vagrant config for your preferred platform.  We could then adapt it for the mac and any others.

lundmar, 2023-03-03:
You pretty much have the dependencies and build steps described in either of the following two files:

https://github.com/lxi-tools/lxi-tools.snapcraft/blob/master/snap/snapcraft.yaml
https://github.com/lxi-tools/lxi-tools.flatpak/blob/master/io.github.lxi-tools.yaml

I expect similar steps are required in a Vagrant macos VM but instead using brew to install the various dependencies including meson. Then use meson to build e.g. liblxi and lxi-tools etc. as done in the files mentioned.

My question is, is mac os free to download and install in a Vagrant VM or docker container?

lundmar, 2023-03-29:
Closing issue as I'm not familiar with Vagrant and macOS. For lxi-tools to gain macOS it must be driven by someone with access to macOS.

closed 2023-03-29 by lundmar