lxi-tools/lxi-tools #69: Feature request: Rigol MSO4000 Series

MarkSupp, 2023-03-04:
Would it be possible to add the Rigol MSO4000 Series to the list of supported devices? Currently, it's only possible to receive the identifier but no screen dump. I'm willing to help with testing and I am able to provide documentation if needed.
Thanks in Advance!

lundmar, 2023-03-04:
Hi @MarkSupp I've looked at the programmers manual for Rigol MSO4000 and it looks like we can reuse the rigol-2000 screenshot plugin.

You can test it by selecting the plugin manually.

I've made a change so it should automatically detect and select the right plugin. It will soon be available on the snap edge channel.

MarkSupp, 2023-03-06:
Thank you very much, I am eager to try it out!

closed 2023-03-06 by MarkSupp

MarkSupp, 2023-03-06:
I'm passing along the requests of my father, since he is not used to working with GitHub:
Not tested yet - waiting for latest Snap Edge version for it.
My systems are MSO4024 and MSO4054.
You could extend the plugin also to the DS4000 series - they are the same, despite the logic probe.

lundmar, 2023-03-06:
Thanks. Let me know if they work then I will update the list of supported instruments.

I'll extend the plugin to work with DS4000 also.

MarkSupp, 2023-03-06:
Latest Snap Edge showed up at my Ubuntu - it works like a charm. Thank you so much for your ultra-fast response.
[image]

lundmar, 2023-03-06:
Great. Thanks for reporting and testing. I hope you find lxi-tools useful.

I've added the two models to the list of supported instruments.

MarkSupp, 2023-03-06:
Hi,
There seems to be a Problem with the Lua script not recognizing certain keywords/commands. Do you possibly have an idea why this might be happening? Some commands (print, tonumber) seem to work while others (connect, msleep, scpi) do not.

lundmar, 2023-03-06:
Yes, in lxi-tools v2.4 the following change was introduced:
https://github.com/lxi-tools/lxi-tools/releases/tag/v2.4

So you must be trying to run some old scripts which have not been updated to use the new API.

Examples and doc have been updated:
https://github.com/lxi-tools/lxi-tools/blob/master/doc/lua-api.txt
https://github.com/lxi-tools/lxi-tools/tree/master/examples/lua

MarkSupp, 2023-03-06:
On behalf of my father and I we can't thank you enough for your help! As a sidenote, it might also be a good idea to change the examples given in the Readme on the front page.

lundmar, 2023-03-06:
You're welcome. Yes, I need to update the screenshots.