lxi-tools/lxi-tools #70: Icon missing

dwarning, 2023-03-19:
I have a snap lxi-gui installation on Ubuntu 22.04 - version 2.6. Everything working fine. The (small) problem is, that after a automatic snap update the icon in the application panel is gone.  Only text is shown _LXI Tools_ and I can start the application.
It seems something has changed because before the icon was there.

lundmar, 2023-03-19:
Hmm, my application icon on the desktop is not gone but it does not show in the "about" dialog. I'll look into it.

dwarning, 2023-03-20:
I am really not sure if it is a bug. After a general Ubuntu actualization and new start the icon is back. Sorry for confusion - may be you can close the issue.
For clarification: The missing icon was in the Ubuntu application center not on my desktop. The icon in my "about" dialog is also missing or destroyed.
Thank you for taking care.

lundmar, 2023-03-20:
No worries. I'll close the issue. Either way, I need to fix the about dialog icon.

closed 2023-03-20 by lundmar