lxi-tools/lxi-tools #71: Can't connect to ssA 3032x

XavierExceenis, 2023-05-11:
Hello,

When I try to connect to my SSA 3032x device I can't get access.

lxi discover 
Searching for LXI devices - please wait...

Broadcasting on interface lo
Broadcasting on interface wlp4s0
Broadcasting on interface docker0

No devices found

lundmar, 2023-05-11:
I see in our list of tested instruments that is has been reported to work with lxi+tools, including the discovery feature.

Lxi-tools support two different discovery methods. Have you tried use the nDNS discovery?

Are you running lxi-tools in a VM? In that cause you must remember to bridge the VM network interface with the host, else the broadcast messages used in the discovery mechanism do not reach the instrument.

XavierExceenis, 2023-05-11:
Indeed I try with ethernet connexion on computer and it works properly.
With my computer on Wifi connexion on the same network as the device, the device has a IP but when I try to get screenshoot, it is written, the library can't connect to the device.

Is there a way to connect with Wifi on my computer to the device connected with ethernet on the same network?

lundmar, 2023-05-11:
Ok. Well, it sounds like you have maybe a wrongly configured wireless wifi router. It should provide you access to your local ethernet. If it is blocking both discovery and you can't take screenshot or send scpi messages directly then something is wrong. You will have to check the configuration of your wifi router to make sure it is not blocking the network in some way.

XavierExceenis, 2023-05-11:
An other solution for me could be a direct connexion without network. With a ethernet cable between the device and my computer. Could it be possible ?  I use Ubuntu OS

lundmar, 2023-05-11:
Yes, but your instrument probably need a dhcp server to be assigned an IP. I believe you go into your Ubunto network settings and share you connection with other computers. I think that should make your Ubuntu act as dhcp server for your instrument on the network interface. You will have to figure out the details.

lundmar, 2023-05-13:
I'm going to close this issue as it is a local network configuration issue. Instrument and tool works fine.

closed 2023-05-13 by lundmar