lxi-tools/lxi-tools #72: macOS install

robcarruthers, 2023-07-23:
I've managed to get LXI-Tools to install on macOS Ventura 13.5 (GUI disabled), from source and with a brew formula, by simply adding lua as a dependency.

I think enabling the GUI will come together also, however I currently getting this error when the install runs `postinstall.py`
```shell
❯ HOMEBREW_NO_INSTALL_FROM_API=1 brew install lxi-tools

==> Fetching lxi-tools
==> Downloading https://github.com/lxi-tools/lxi-tools/archive/refs/tags/v2.5.tar.gz
Already downloaded: /Users/rob/Library/Caches/Homebrew/downloads/7b774ec1dbd550cb2353654c4709d7c413439ebabdb1f138c134593ff35b2890--lxi-tools-2.5.tar.gz
==> meson setup build --prefix=/opt/homebrew/Cellar/lxi-tools/2.5 --libdir=/opt/homebrew/Cellar/lxi-tools/2.5/lib --buildtype=release --wrap-mode=nofallback
==> meson compile -C build --verbose
==> meson install -C build
Last 15 lines from /Users/rob/Library/Logs/Homebrew/lxi-tools/03.meson:
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'gtk-update-icon-cache'
Installing src/lxi to /opt/homebrew/Cellar/lxi-tools/2.5/bin
Installing src/lxi-gui to /opt/homebrew/Cellar/lxi-tools/2.5/bin
Installing /private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/build/man/lxi.1 to /opt/homebrew/Cellar/lxi-tools/2.5/share/man/man1
Installing /private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/data/io.github.lxi-tools.lxi-gui.desktop to /opt/homebrew/Cellar/lxi-tools/2.5/share/applications
Installing /private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/data/io.github.lxi-tools.lxi-gui.appdata.xml to /opt/homebrew/Cellar/lxi-tools/2.5/share/metainfo
Installing /private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/data/io.github.lxi-tools.lxi-gui.gschema.xml to /opt/homebrew/Cellar/lxi-tools/2.5/share/glib-2.0/schemas
Installing /private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/data/icons/hicolor/scalable/apps/io.github.lxi-tools.lxi-gui.svg to /opt/homebrew/Cellar/lxi-tools/2.5/share/icons/hicolor/scalable/apps
Installing /private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/data/icons/hicolor/symbolic/apps/io.github.lxi-tools.lxi-gui-symbolic.svg to /opt/homebrew/Cellar/lxi-tools/2.5/share/icons/hicolor/symbolic/apps
Running custom install script '/private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/build-aux/meson/postinstall.py'
FAILED: install script '/private/tmp/lxi-tools-20230723-22992-165tvli/lxi-tools-2.5/build-aux/meson/postinstall.py' exit code 1, stopped
```

I suspect the script is just looking in the wrong place (or rather brew has put `gtk-update-icon-cache` in a different place).

An explanation of what this script does and requires would help and be much appreciated.

lundmar, 2023-07-23:
The script calls some tools to update various caches that need to know about your newly installed application, such as the icon cache, and gsettings (registry) cache.

In the post install script, these tools are not called when DESTDIR is set. DESTDIR is typically set if you are creating a package and installing to a package directory which content will later be installed in a system via a package manager which is then responsible for calling these tools.

In the case of homebrew it looks like it does not deal in packages but simply installs directly in the system and so DESTDIR is not set so the way to solve the build issue would be to define some build dependencies to the homebrew formulas that provide these tools.

As far as I know gtk-update-icon-cache is a tool provided by gtk4. I guess update-desktop-database is provided by the desktop-file-utils and glib-compile-schemas is provided by gsettings-desktop-schemas.

Make sure you include these dependencies. If the tools can not be found then there may be some path configuration issue.

robcarruthers, 2023-07-23:
Thanks for that. GPT recommended trying gtk-3 rather than 4, to get past the icon-cache error and it seems to work with the dependencies below in the brew formula.

Unless you can identify any issues, I'll submit the formula to homebrew and see if it passes their CI.

```ruby
  depends_on 'lua'
  depends_on 'liblxi'
  depends_on 'glib'
  depends_on 'gtksourceview5'
  depends_on "gtk+3"
  depends_on 'desktop-file-utils'
  depends_on 'libadwaita'
  depends_on 'json-glib'
  depends_on "meson" => :build
  depends_on "ninja" => :build
```

[image]

lundmar, 2023-07-23:
Great, nice to see lxi-gui running on mac! :)

I investigated the icon cache issue and it turns out that the post install script should really be calling gtk4-update-icon-cache instead gtk-update-icon-cache (gtk3). I've fixed the post install script in https://github.com/lxi-tools/lxi-tools/commit/0d6c4eda655750834b563b7049a8efcd41ec5ee9.

You can simply apply that patch in your formula and depend on gtk4 instead of gtk3 and I think it will work and we will have a proper dependency definition.

robcarruthers, 2023-07-23:
That works, with the formula pointing to master, ie:
```shell
brew install lxi-tools --HEAD
```
I'll push the updated formula now and leave the issue open until homebrew have merged it.

Whats the timing for the next release?

Thanks again for these tools, the CLI and scripting capabilities are a god send!

lundmar, 2023-07-23:
You may want to lock it to the specific commit or introduce a patch to the official version. I don't know what is normal practice in homebrew. Following HEAD is bad idea as it can break.

I'm not sure when a new release will happen - I think we need to gather some more changes to make a release. When I get time I may look into adapting the latest UI features that libadwaita offers to offer some more polish to the GUI.

No worries, and thank you for your mDNS Bonjour contribution and making the tools work fully on macOS. If you have ideas to improve lxi-tools feel free to make suggestions or even code pull request :)

Oh, by the way, I plan to kick off a new lxi-tools related project in https://github.com/lxi-tools/lxi-tools-scripts - a repository intended for lxi-tools lua scripts for various test instruments. It is very loosely defined right now but the idea is to have a collection of lua scripts that provide the same API across various test instruments. For example, for DC power supply scripts, instead of working directly with obscure SCPI command to read the voltage of channel 1, these scripts would instead offer the same functionality via a simplified lua API, e.g. ```voltage = get_voltage_channel(1)```.

Anyway, in case you use the lxi-tools scripting feature you are welcome to start building up such scripts for the instruments that you use.

robcarruthers, 2023-07-23:
`brew install lxi-tools --HEAD` simply builds from source on the formula specified branch, which is master currently. So in the interim I think it should be fine.

I would welcome any sort of standardization around instrument communication and categorization and be happy to contribute scripts for the instruments we install regularly.

lundmar, 2023-07-23:
Exactly, a common lua API for various classes of test instruments would be useful. Feel free to make a PR on the repo for any scripts you may want to add. We can figure out the API as we go along. When I get some time I plan to start adding script support for my R&S NGM202 DC power supply.

robcarruthers, 2023-07-23:
That sounds great, once you've created a starting point for the R&S PSU, we could add others. I guess PSUs and multi-meters would be the low hanging fruit initially, I could look at a multi-meter API once I see the direction you're heading in.

lundmar, 2023-07-23:
Oh, by the way, let me know when lxi-tools is officially available via homebrew. Then we can update the install instructions.

robcarruthers, 2023-07-25:
It looks like homebrew won't accept [the formula](https://github.com/Homebrew/homebrew-core/pull/137339) until the gtk-update-icon-cache fix is released. So I guess there's no hurry, I'm ready to go when you are.

Regarding an Instrument API, that sounds great, once you've created a starting point for the R&S PSU, we could add others. I guess PSUs and multi-meters would be the low hanging fruit initially, I could look at a multi-meter API once I see the direction you're heading in.

lundmar, 2023-07-25:
Right. I'll see if I can find time to make a new release this upcoming weekend.

I'm currently quite busy working for a client but that project ends in about a month so after that I will have time to dig into the lxi-tools-scripts repo and create a small framework with some sensible simple API for DC power supply to begin with.

hooperbloob, 2023-07-25:
I'm so looking forward to being able to build & test the gui on MacOS, thanks guys.

> create a small framework with some sensible simple API for DC power supply to 

If/when that comes to be I hope it can be as parametric as possible. Incorporating new instruments via json config files would greatly expand the ecosystem by non-developers.

lundmar, 2023-07-28:
@robcarruthers FYI - I've released lxi-tools v2.6 so you can move on with the homebrew formula.

hooperbloob, 2023-08-10:
I just tried installing it via   'brew install lxi-tools'.  It seemed to install successfully, the lxi cli seems to work, 
it picks up instruments on my network.  Trying to launch the lxi-gui it looks like something is missing:

brack@WASML bin % pwd
/usr/local/Cellar/lxi-tools/2.6/bin

brack@WASML bin % ls -l
total 712
-r-xr-xr-x  1 brack  admin   73864 Jul 27 19:09 lxi
-r-xr-xr-x  1 brack  admin  285112 Jul 27 19:09 lxi-gui

brack@WASML bin % lxi-gui               
(lxi-gui:31792): GLib-GIO-ERROR **: 00:04:10.085: Settings schema 'io.github.lxi-tools.lxi-gui' is not installed

Also, is there something that could be added to the install script to add it as an entry in my applications folder?

lundmar, 2023-08-10:
@hooperbloob Can you check that brew installed the file `io.github.lxi-tools.lxi-gui.gschema.xml` ?

hooperbloob, 2023-08-10:
Doesn't look like it:

bin % gsettings list-schemas
org.gnome.desktop.a11y
org.gnome.desktop.a11y.applications
org.gnome.desktop.a11y.interface
org.gnome.desktop.a11y.keyboard
org.gnome.desktop.a11y.magnifier
org.gnome.desktop.a11y.mouse
org.gnome.desktop.app-folders
org.gnome.desktop.background
org.gnome.desktop.calendar
org.gnome.desktop.datetime
org.gnome.desktop.default-applications
org.gnome.desktop.default-applications.office
org.gnome.desktop.default-applications.office.calendar
org.gnome.desktop.default-applications.office.tasks
org.gnome.desktop.default-applications.terminal
org.gnome.desktop.input-sources
org.gnome.desktop.interface
org.gnome.desktop.lockdown
org.gnome.desktop.media-handling
org.gnome.desktop.notifications
org.gnome.desktop.peripherals
org.gnome.desktop.peripherals.keyboard
org.gnome.desktop.peripherals.mouse
org.gnome.desktop.peripherals.pointingstick
org.gnome.desktop.peripherals.touchpad
org.gnome.desktop.peripherals.trackball
org.gnome.desktop.privacy
org.gnome.desktop.screensaver
org.gnome.desktop.search-providers
org.gnome.desktop.session
org.gnome.desktop.sound
org.gnome.desktop.thumbnail-cache
org.gnome.desktop.thumbnailers
org.gnome.desktop.wm.keybindings
org.gnome.desktop.wm.preferences
org.gnome.system.locale
org.gnome.system.location
org.gnome.system.proxy
org.gnome.system.proxy.ftp
org.gnome.system.proxy.http
org.gnome.system.proxy.https
org.gnome.system.proxy.socks
org.gtk.Demo
org.gtk.Demo4
org.gtk.Settings.ColorChooser
org.gtk.Settings.Debug
org.gtk.Settings.EmojiChooser
org.gtk.Settings.FileChooser
org.gtk.exampleapp
org.gtk.gtk4.Settings.ColorChooser
org.gtk.gtk4.Settings.Debug
org.gtk.gtk4.Settings.EmojiChooser
org.gtk.gtk4.Settings.FileChooser

lundmar, 2023-08-10:
Yes, but I need to know if the .xml has been installed. What you are listing there is the entries from the generated gsettings cache which is generated from the .xml files. Meaning, the problem may be that either the .xml has not been installed or the .xml has been installed but the cache has not been regenerated.

lundmar, 2023-08-10:
lxi-tools should install the following files:
```
.
├── bin
│   ├── lxi
│   └── lxi-gui
└── share
    ├── applications
    │   ├── io.github.lxi-tools.lxi-gui.desktop
    │   └── mimeinfo.cache
    ├── bash-completion
    │   └── completions
    │       └── lxi
    ├── glib-2.0
    │   └── schemas
    │       ├── gschemas.compiled
    │       └── io.github.lxi-tools.lxi-gui.gschema.xml
    ├── icons
    │   └── hicolor
    │       ├── icon-theme.cache
    │       ├── scalable
    │       │   └── apps
    │       │       └── io.github.lxi-tools.lxi-gui.svg
    │       └── symbolic
    │           └── apps
    │               └── io.github.lxi-tools.lxi-gui-symbolic.svg
    ├── man
    │   └── man1
    │       └── lxi.1
    └── metainfo
        └── io.github.lxi-tools.lxi-gui.appdata.xml

```

hooperbloob, 2023-08-10:
Ah, this is what I have:

[image]

Looks like gschemas.compiled is missing from mine.

lundmar, 2023-08-10:
Right. I know Rob had some issues with the gchemas.compiled file - the brew build system complained about it because of some conflict. To solve that it was simply deleted but now it turns out we need it. I think we need the brew maintainers give us a hint how to solve this.

robcarruthers, 2023-08-12:
Sorry about this, I've managed to replicate the error locally, but don't have a solution just yet.

We [removed the gchemas.compiled file](https://github.com/Homebrew/homebrew-core/pull/137867#issuecomment-1659400236) to avoid an error in the brew CI and in the absence of any clear direction introduced the error below
```
❯ lxi-gui

(lxi-gui:48083): GLib-GIO-ERROR **: 14:19:08.379: Settings schema 'io.github.lxi-tools.lxi-gui' is not installed
fish: Job 1, 'lxi-gui' terminated by signal SIGTRAP (Trace or breakpoint trap)
```

If I edit the formula to leave the file in place and run the formula locally...
```
==> Fetching lxi-tools
==> Downloading https://github.com/lxi-tools/lxi-tools/archive/refs/tags/v2.6.tar.gz
Already downloaded: /Users/rob/Library/Caches/Homebrew/downloads/2b1b5b1194c0457cc36054988e392a269b59a1a2581f351eee93bd0dfaa84c35--lxi-tools-2.6.tar.gz
==> Reinstalling lxi-tools 
==> meson setup build
==> meson compile -C build --verbose
==> meson install -C build
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /opt/homebrew
Could not symlink share/glib-2.0/schemas/gschemas.compiled
Target /opt/homebrew/share/glib-2.0/schemas/gschemas.compiled
already exists. You may want to remove it:
  rm '/opt/homebrew/share/glib-2.0/schemas/gschemas.compiled'

To force the link and overwrite all conflicting files:
  brew link --overwrite lxi-tools

To list all files that would be deleted:
  brew link --overwrite --dry-run lxi-tools
```
If I do this and overwrite the linking, everthing works as expected. So now I'm looking for a way to force the link overwrite in the formula.

If no one has any other ideas, I'll submit a formula patch and and ask the question of the homebrew maintainers.

lundmar, 2023-08-12:
From hooperbloobs screenshot it looks like brew installs applications in their own directory tree and not in a global tree.

This is very similar to the way that I install lxi-tools locally in my home directory when I develop and test the application. When I do that I need to use the following environment to start the application eg.:

```
LD_PRELOAD=$HOME/opt/liblxi/lib/x86_64-linux-gnu/liblxi.so LD_LIBRARY_PATH=$HOME/opt/lxi-tools/lib/x86_64-linux-gnu XDG_DATA_DIRS=$HOME/opt/lxi-tools/share:$XDG_DATA_DIRS ./opt/lxi-tools/bin/lxi-gui
```

I think brew does something similar so when it runs lxi-gui it sets the XDG_DATA_DIRS variable accordingly so that gsettings will find the locally installed gschemas.compiled file. Se we definitely need to install the gschemas.compiled file.

However, the error:
```
rm '/opt/homebrew/share/glib-2.0/schemas/gschemas.compiled'
```
indicates there is also a global homebrew directory tree in which brew maintains a gschemas.compiled file and for some reason that conflicts during install on the CI server.

Maybe they install things differently on the CI server. Maybe on the CI server they install things in a global directory tree and thus the conflict when it tries to overwrite an existing gschemas.compiled file.

Either way, this is a very brew tool specific issue that the brew guys must be able to answer.

lundmar, 2023-08-12:
The only related issue I can find in the homebrew issue tracker is https://github.com/Homebrew/homebrew-core/pull/11023 where they hint the solution that something should be done in postinstall but details are lacking.

lundmar, 2023-08-12:
You may solve the issue by regenerating the gschemas file manually in the postinstall step exactly as they do here:

https://github.com/Homebrew/homebrew-core/blob/ecd98b130638d78b86418e6af6e029a6aab4e309/Formula/ghex.rb#L42

I think you should also include the line updating the icon cache and add ```depends_on "hicolor-icon-theme"``` - that may fix any icon issue.

robcarruthers, 2023-08-13:
After reading [this post](https://askubuntu.com/questions/1346576/how-to-fix-glib-gio-error-settings-schema-org-gtk-settings-filechooser-is-not) I blindly installed gtk4 and this seems to fix the error... at least for me, I'm now running the beta version of Sonoma macOS 14
```
brew uninstall lxi-tools
brew install lxi-tools
brew install gtk4
lxi-gui
```

lundmar, 2023-08-13:
I think the only reason that fixes the problem is that you are reinstalling gtk4 and that triggers a regeneration of the gschemas cache. I think the proper solution is to do as the do in the other gtk4 formulas such as the ghex one that I mentioned.

robcarruthers, 2023-08-13:
Thanks, that makes perfect sense.

I think I have a working solution, but lets back track a little first and confirm my logic.
- In the original formula, we added `m_f "#{share}/glib-2.0/schemas/gschemas.compiled"` to correct the linking error that occured in CI
- This got us past the CI and to a merged formula, however it caused the error that hooperbloob reported with the `lxi-gui` command

Sometimes, I can't see the wood for the trees...

If I use [link_overwrite](https://rubydoc.brew.sh/Formula.html#link_overwrite-class_method) in the correct location as below, we fix both these issues
```
  depends_on "desktop-file-utils"
  depends_on "gtksourceview5"
  depends_on "json-glib"
  depends_on "libadwaita"
  depends_on "liblxi"
  depends_on "lua"
  depends_on "hicolor-icon-theme"

  link_overwrite "share/glib-2.0/schemas/gschemas.compiled"

  def install
    system "meson", "setup", "build", *std_meson_args
    system "meson", "compile", "-C", "build", "--verbose"
    system "meson", "install", "-C", "build"

    rm_f "#{share}/glib-2.0/schemas/gschemas.compiled"
  end

  def post_install
    system "#{Formula["glib"].opt_bin}/glib-compile-schemas"
    system "#{Formula["gtk4"].opt_bin}/gtk4-update-icon-cache", "-f", "-t", "#{HOMEBREW_PREFIX}/share/icons/hicolor"
  end
```
The only thing I'm not sure about is the hard coded path, the docs don't say but [all examples](https://github.com/Homebrew/homebrew-core/blob/master/Formula/glib.rb) I can see, seem to be relative to the formula working directory.

lundmar, 2023-08-13:
Your glib-compile-schemas call is broken, it should look like this:
```
system "#{Formula["glib"].opt_bin}/glib-compile-schemas", "#{HOMEBREW_PREFIX}/share/glib-2.0/schemas"
```
Else it will not generate the gschemas.compiled in the local lxi-tools install directory.

I don't think you need the link_overwrite stuff because we delete the generated cache file in the install step but generate it in postinstall where I expect it does not do the same link check.

lundmar, 2023-08-16:
@robcarruthers I think all you need to do is not delete the bottle section in your commit.

robcarruthers, 2023-08-16:
Yes, thanks. I submitted [a revision](https://github.com/Homebrew/homebrew-core/pull/139482) to the formula as suggested. It wouldn't install locally with the bottle block so I removed it.

However, the maintainers have said it needs to be present. I think it will progress, but it's waiting on their clarification now...

lundmar, 2023-08-19:
Thanks for reporting. We are aware that there are issues with the brew package regarding lxi-gui. An updated brew package should soon be available which fixes these issue.

lundmar, 2023-08-19:
I just noticed that the window tab icons does not look as crispy as they should (png vs svg). I hope the icon cache handling in the new Homebrew package fixes that.
[image]

robcarruthers, 2023-08-20:
I think for now the original issue has been solved, or at least will be once the revised formula is merged (Still working on this). So I'll close the issue and open new one for the interface discrepancies.

closed 2023-08-20 by robcarruthers

lundmar, 2023-08-20:
@robcarruthers Regarding the current formula it looks like you merged the wrong way:

```Merge branch 'master' of https://github.com/Homebrew/homebrew-core into lxi-tools-glib-fix```

In this situation, merges are normally only done the other way, ie. merge of your branch into master. In case of github PR, this is done by the github system if needed when it merges the final PR.

Instead of a merge you should have done ```git rebase master``` while standing on your branch and then maybe a force push. This basically takes your branch with the two commits and moves it forward so the branch branches out from the latest master. Rebasing is a little more exotic git concept that takes a little practice to get used to but generally when you think you need to merge you often really want to rebase :)

I'm not sure how to fix your current PR branch. Now that I have released lxi-tools v2.7 it may just be easier to create a new PR with v2.7 including all the changes.

robcarruthers, 2023-08-22:
Finally, the [formula](https://github.com/Homebrew/homebrew-core/pull/139973) has been merged, incorporating v2.7
```
brew install lxi-tools
```

lundmar, 2023-08-22:
That's great. Is there any way you guys can test that the original installation issue is gone? I mean, maybe uninstall everything including gtk3 etc. and then reinstall lxi-tools to make sure it does not need any of the tricks mentioned earlier.

robcarruthers, 2023-08-23:
I've exercised the interface and cli were pretty thoroughly. Including removing deps and reinstalling, tab and window functions/operations, settings and about panels, cli discover, mdns etc.

It all looks pretty good.

lundmar, 2023-08-23:
@robcarruthers Great. I was mostly worried about the dep thing and glib error. Thanks for fixing the formula 👍🏻