lxi-tools/lxi-tools #73: Interface font size macOS

robcarruthers, 2023-08-20:
There appears to be differences between the [macOS interface](https://github.com/lxi-tools/lxi-tools/issues/72#issuecomment-1685042575) font size and linux, see images below, Linux is in the background. It appears to effect all text and icons.

[image]

As best as I can tell, both interfaces seem to be using vector graphics (Its hard to screen capture this).

Is this something we need to be concerned with?
If so, how can we change the font size for macOS to be more inline with the Linux experience.

There is also a console error when right clicking on the interface.
```
(lxi-gui:62926): Gtk-WARNING **: 14:04:19.005: Broken accounting of active state for widget 0x142c091c0(GtkPopoverMenu)
```
I'd guess this is a seperate issue and I have no idea what it means.

lundmar, 2023-08-20:
I'm not gonna make a big deal out of this issue because Homebrews build system seems a bit quirky to me. However, I did find an example brew recipe with the following two lines I think fixes the issue in the post install step: https://github.com/Homebrew/homebrew-core/blob/7b04a0ca616afd729f070bd3d0d2c8cc32a20f52/Formula/lib/librsvg.rb#L60

Basically the problem is that GTK uses certain image loaders (gdk-pixbuf loaders) to handle different image formats such as .png, .svg, etc. but it fails to find the svg loader because it is not correctly defined in the local loader.cache file of the application. Because the svg loader is not loaded it falls back to using the png version of the icons and they do not scale well, hence the discrepancy in presentation.

lundmar, 2023-08-20:
Oh, by the way, we can ignore the console warning. GTK is know to spew out random warnings like that depending what GTK version is used.

lundmar, 2023-08-20:
Actually, I just realize that the solution may be to simply add ```depends_on "librsvg"``` as it seems to install the svg gdk loader. If that does not work, you may need to add the mentioned lines in the post install step.

If nothing works, don't bother.

robcarruthers, 2023-08-20:
I've tried all the suggestions with no luck. I've also exercised the interface pretty thoroughly are can't find any other issues other than the font size, which does effect all the buttons unfortunately.

I've submitted a [new PR](https://github.com/Homebrew/homebrew-core/pull/139973) which bumps the formula for the new release and incorporates the changes from #72, fingers crossed!

lundmar, 2023-08-20:
Ok. Well, let's just leave it - it is just cosmetics. The most important thing is that the features work.

closed 2023-08-22 by robcarruthers