lxi-tools/lxi-tools #76: Siglent SDS1104X-E not being discovered

sethkaz, 2023-10-02:
I have a Siglent SDS1104X-E scope (and X-U, but not working on that one yet).

I can't seem to get the scope to be discovered by the search tool.  This is in both the command line and the gui.  However, when I manually add the scope, I get SCPI responses and screenshots working as expected.  

Is there something to be done to set it to be discoverable?  Either in the scope settings or lxi-tools.

lundmar, 2023-10-02:
@sethkaz We know that SDS1204X-E works and since your instrument is part of that same family I expect it to be discoverable too. Likely it has something to do with your network configuration. For example, if you are running lxi-tools in a VM you need to make sure that the network device of your VM is in bridge mode so that the discovery mechanism is able to broadcast its discovery messages across your network interface.

sethkaz, 2023-10-02:
@lundmar Thanks for the prompt response.  I saw that detail and was expecting the same.  I'm currently not running it in a VM.

What are the discovery message types? So I can check my router to see if it's actually blocking those or if something else is up.

Also, I'd be happy to contribute to testing new equipment and writing up docs, if that would helpful.

lundmar, 2023-10-02:
Assuming you are using the regular discovery mechanism (not mDNS) you should look for something like this:

[image]

In this case a VXI-11 discovery request is sent to broadcast address 192.168.0.255 on port 111. A response is sent back from 192.168.0.107.

You could also try discovery your instrument via mDNS setting but I don't remember if the SDS series supports that.

All types of contributions are welcome. Testing and expanding our list of tested instruments is always helpful. There is also our wiki where one is free to elaborate on their experience or programming details with various instruments.

lundmar, 2023-10-16:
@sethkaz Any luck solving the issue?