lxi-tools/lxi-tools #79: Crashes when instrument is not found on script screen.

crow0827, 2023-12-19:
With following example code, crashes occur when instrument is not found on script screen.
If I enter the correct IP address, crash not occurred.
Isn't it bug?

OS: Ubuntu 22.04
Ver: 2.7

```
-------------------------------------
--  lxi-tools                      --
--    https://lxi-tools.github.io  --
-------------------------------------

-- Example: Retrieve ID from instrument (R&S NGM 202) via VXI11 protocol

-- Connect to power supply
psu = lxi_connect("192.168.0.107", nil, nil, 2000, "VXI11")

-- Print instrument ID
print("Power Supply ID = " .. lxi_scpi(psu,"*IDN?\n"))

-- Disconnect
lxi_disconnect(psu)
```

lundmar, 2023-12-24:
Can you paste the exact crash error that you see?

crow0827, 2023-12-24:
If an exception occurs in Lua's operating environment, does it normally generate an error statement on the GUI?
Nothing is displayed, so I do not know what error actually occurred.

lundmar, 2023-12-24:
If you start lxi-gui from a console it may print some lua crash information in the console.

lundmar, 2023-12-24:
I've just tested your code in my end using the snap version. The lxi-gui application does not crash but simply returns the following in the lxi-gui output window:
```
Power Supply ID = -1.0
```

crow0827, 2023-12-24:
After trying, the error cause is lxi_disconnect(psu).  If no device is able to connect, "psu" is -1.
With lxi-gui console, "Segmentation fault(Core dumped)" error info is outputted.

lundmar, 2023-12-24:
Hmm, what version of lxi-tools are you running?

Did you install lxi-tools via a native distro package or are you running it via snap?

crow0827, 2023-12-24:
Ver 2.7 via snap.

lundmar, 2023-12-24:
When I run the snap installed version I get the following in the terminal output window:

```
lundmar@wopr:~$ snap run lxi-tools.lxi-gui
Gsk-Message: 13:20:01.477: Failed to realize renderer of type 'GskGLRenderer' for surface 'GdkX11Toplevel': Failed to create EGL display

Error: Failed to connect
Error: Failed to send message

```

No segfault.

I'm testing on Ubuntu 23.10.

As far as I know everything that lxi-gui relies on should be installed in the snap so we should be using the same version of everything, including the version of lua, liblxi etc.

This makes the problem a bit puzzling.