lxi-tools/lxi-tools #83: report  AimTTi CPX400DP

dbozec, 2024-01-26:
Hi, just got a brand new CPX400DP Power Supply with LXI capability, but I can't discover it on network.
Still I can connect directly,
```bash
$ lxi scpi -a 192.168.1.103 -i
Connected to 192.168.1.103
Entering interactive mode (ctrl-d to quit)

lxi>
```
but I got errors on SCPI commands
```
$ lxi scpi -a 192.168.1.100 "*IDN?"
Error: Unable to connect to LXI device
```
However SCPI Commands work as expected from the web interface,
"*IDN?" returns "THURLBY THANDAR, CPX400DP, 581316, 3.00-4.12"

MichaelChamberlain79, 2024-01-26:
Couldn't get the auto network discovery working either. From what I gathered it seems the AimTTI doesn't conform to some spec mDNS (I think). Just set the PSU to a static IP on your network (from web interface or router)

I'm using an AimTTI PSU, but the only way to get it working is to specify the port number too.

e.g to read the Voltage from CH1:

lxi scpi -a 192.168.1.103 -p 9221 -r "V1?"

dbozec, 2024-01-26:
Indeed it works when specifying the port !
Thank you !

lundmar, 2024-01-26:
It is puzzling that these instruments do not seem to implement discovery via VXI11/mDNS, it is part of the LXI standard.

lundmar, 2024-01-26:
I've added the instrument to the list of tested instruments. Thanks guys.

closed 2024-01-26 by lundmar

MichaelChamberlain79, 2024-01-26:
Martin, I'm also using AimTTI PL303-P successfully!

Setting Voltage and Current and reading the values again (from Raspberry PI 5 running Raspbian and lxi-tools)

lundmar, 2024-01-26:
@MichaelChamberlain79 Also added now. Thanks 👍🏻