lxi-tools/lxi-tools #84: Only finds _lxi._tcp services on MacOS, not the others

u77345, 2024-02-11:
When running lxi v2.7 on MacOS, installed by brew, the mdns discovery only finds the `_lxi._tcp` registrations, not the other expected ones:

```
'_hislip._tcp' 
'_scpi-telnet._tcp'
'_vxi-11._tcp'
'_scpi-raw._tcp'
```

lundmar, 2024-03-27:
Sorry for the late response - I've been busy IRL.

Can you please try search using the command line tool?

When searching via mdns on Linux it works fine:

```
lundmar@wopr:~$ lxi discover --mdns
Searching for LXI devices - please wait...

  Found "Rohde&Schwarz power supply NGM202-101403" on address 192.168.0.107
    lxi service on port 80
  Found "RS Power Supply NGL202-103990" on address 192.168.0.176
    lxi service on port 80
  Found "Rohde&Schwarz power supply NGM202-101403" on address 192.168.0.107
    scpi-raw service on port 5025
  Found "RS Power Supply NGL202-103990" on address 192.168.0.176
    scpi-raw service on port 5025

Found 4 services
```

It should work the exact same on Mac but unfortunately I don't have access to a Mac for testing.

lundmar, 2024-09-29:
Fixed in liblxi here https://github.com/lxi-tools/liblxi/pull/49

closed 2024-09-29 by lundmar