lxi-tools/lxi-tools #86: Is it somehow possible to extend the IP address to include LAN Device name ?

PavelKCZ, 2024-02-25:
I have a HP34401A multimeter with attached Raspberry Pi Zero and GPIB interface. The multimeter GPIB address is 22.
When I want to communicate with this instrument via Labview VISA interface, I am using the address string in form "TCPIP0::192.168.1.xyz::gpib,22:INSTR" and everything in Labview works OK. The "gpib,22" is LAN Device name.

But how I can modify the lxi-gui to such kind of IP address extension ? If I send the *IDN? command only to the address 192.168.1.xyz, I get back the answer "python-vxi11-server,bbb,123,567" which an answer from the python vxi server itself, not from the instrument.

lundmar, 2024-02-25:
Your vxi server should relay any requests to your gpib device via VXI11 protocol on port 111. 

Which gpib interface are you using?

PavelKCZ, 2024-02-25:
Hello,

I am using gpib-bitbang driver with this board:

https://www.hackster.io/lightside-instruments/the-gpib4pi-gpib-for-raspberry-pi-shield-4b3e9a
https://www.hackster.io/lightside-instruments/wireless-lan-gpib-gateway-with-open-source-hardware-6e0af8  (just some more python software, for wireless access)

From Labview, everything works OK.

Also, everything works fine, if I try simple python script from Ubuntu machine on my network:

import vxi11
instr = vxi11.Instrument("TCPIP::192.168.1.164::gpib,22::INSTR")
instr.write("*IDN?")
print(instr.read())

lundmar, 2024-02-25:
I see. No, unfortunately right now the lxi-gui does not support specifying the VXI11 device name in the connect dialog. I'll add this to the TODO list for next release. Currently, the VXI11 device name defaults to 'inst0' which is normally supported by test instruments. So the only way for you to connect via the lxi-gui for now would be to try change the device name of your server from 'gpib,22' to 'inst0'.

That being said, the VXI11 device name is supported and exposed via the lxi-tools script interface. Meaning something like this should work:

```
-------------------------------------
--  lxi-tools                      --
--    https://lxi-tools.github.io  --
-------------------------------------

-- Example: Retrieve ID from instrument via VXI11 protocol

-- Connect to instrument
psu = lxi_connect("192.168.1.164", nil, "gpib,22", 2000, "VXI11")

-- Print instrument ID
print("Power Supply ID = " .. lxi_scpi(psu,"*IDN?\n"))

-- Disconnect
lxi_disconnect(psu)
```

PavelKCZ, 2024-02-25:
Perfect. I will simply wait for a next release, no problem.
Thank you very much for your assistance.
Have a nice day

PavelKCZ, 2024-02-25:
I just tried from Lua, as you suggested, and it works perfectly. :)
Answer is "HEWLETT-PACKARD,34401A,0,10-5-2" which is correct.

lundmar, 2024-02-25:
Thanks for testing that it works. I'll make sure to expand the connection dialog with this feature in the next release.

closed 2024-02-25 by lundmar