lxi-tools/lxi-tools #90: New features  added

psalic, 2024-09-23:
Hi,

It's not an issue. I added new code on my forked repo.
[https://github.com/psalic/lxi-tools](https://github.com/psalic/lxi-tools)

It includes 3 branches with following features:
- 3 plugins for Agilent instruments
- Live view support
- Windows native support. Build based on MSYS2

Regards, Primoz

lundmar, 2024-09-23:
Hi @psalic,

I took a quick look at your branches - nice work.

Question, does your resulting lxi-tools Windows installer require the user to install msys2 or is it a fully standalone install executable that ships the msys2 libs etc.?

psalic, 2024-09-23:
Hi @lundmar 

Its standalone. No Msys required. All dll, lib, share and etc files are packaged in exe.

lundmar, 2024-09-23:
> Its standalone. No Msys required. All dll, lib, share and etc files are packaged in exe.

Very nice - I think that is what Windows users expect.

I assume your plan is to upstream your changes. Maybe you can start by making a pull request for the agilent plugins.

Regarding the live view feature, I notice you added two buttons, one to start live view and another to stop. Perhaps it would be better UX to consolidated these two buttons into one toggle button which unpressed labels "Live view" and unpressed labels "Stop" and react accordingly.

psalic, 2024-09-23:
I've open pull request.
I will check live view suggestion. Development is needed.

lundmar, 2024-09-23:
I've merged your PR, thanks.

For inspiration you can look at the benchmark toggle button.

psalic, 2024-09-23:
Thank you for tip.
Did you try windows version in VM? I am also linux user and am using VMs. All yob was done for coworkers.. windows users.
It was confirmed it works with new plugins.

lundmar, 2024-09-23:
> Did you try windows version in VM?

Sorry no. I don't actually have a Windows host readily available to test on.

Maybe we can make it a CI job. I believe github CI service features a Windows server but I may remember wrong.

psalic, 2024-09-24:
@lundmar 

I update live_view branch as you suggested

psalic, 2024-09-24:
I also added new branch in liblxi.
[https://github.com/psalic/liblxi](https://github.com/psalic/liblxi)

lundmar, 2024-09-24:
> I update live_view branch as you suggested

Great. I'm a little busy right now but I hope to get some time to test it this weekend.

lundmar, 2024-09-24:
> I think msys2_scripts doesn't belong in this repository. We should have new repository only with msys2_scrips. What is your opinion.

That was actually my first thought when I saw it.

I'll create a repository and grant you write access to it. I hope you will be able to help maintain it because I simply don't have the time nor Windows access to do so myself.

lundmar, 2024-09-24:
I've created https://github.com/lxi-tools/lxi-tools-msys2 and granted you write access. Feel free to push and maintain your m2sys scripts there.

psalic, 2024-09-24:
Hi,

Thank you for repository. I will push my scripts there once sources on lxi-tools and liblxi are ready.  I will do my best to maintain repository.

lundmar, 2024-09-28:
Please see https://github.com/psalic/lxi-tools/commit/703eabf7ee8886790b589c3b33ddf0eda44fabb9#r147326479

lundmar, 2024-09-28:
I've tested your live_view branch and it seems to work well. It survives and restores the buttons correctly even if the instrument suddenly turns off during a live view session etc.

If you fix the code duplication issue I think it will be ready to go in.

psalic, 2024-09-28:
Hi @lundmar ,

I fixed duplicate code on live_view.

Regards, Primoz

lundmar, 2024-09-28:
Great.

Also, please see https://github.com/psalic/lxi-tools/commit/dfaa0c247c4f9b0b20bbe1cbe8da75fe44fdf6db#r147328169

psalic, 2024-09-28:
I dont like plugin_name 32 bytes on stack. I test another change. Minor fix

psalic, 2024-09-28:
@lundmar  Can you please recheck again live_view.

lundmar, 2024-09-28:
Please see https://github.com/psalic/lxi-tools/commit/f13d668c700bd8b516419b0fbd8cf050c7957acc#r147329018

psalic, 2024-09-28:
In mean time I made different fix

psalic, 2024-09-28:
I tried your suggestion. It larger code change than my last commit.

Can you please recheck my commit. I attach you changes based on my last code on live_view branch. Its messier code.
[live_patch.txt](https://github.com/user-attachments/files/17174533/live_patch.txt)

lundmar, 2024-09-28:
Yes, it is a little bit more code but it is better API. I don't want the plugin_name to become an input and output parameter. Doing so means it will always go from plugin auto detection in the first call (when no plugin name is provided) to no auto detection in next calls unless user actively clears the plugin name between calls. I think that is bad API. Hence I would like to keep in an input parameter exclusively.

psalic, 2024-09-28:
OK, will add this diff on branch

lundmar, 2024-09-28:
If you make a pull request I will merge it.

psalic, 2024-09-28:
Pull request open.
I will check conditional compilng for windows now. 
Can you check liblxi on windows branch.
[https://github.com/psalic/liblxi](https://github.com/psalic/liblxi)

lundmar, 2024-09-28:
I think both @robcarruthers and I will be a bit sad if we start interleaving the Bonjour mDNS implementation with cygwin/windows related stuff.

The bonjour.c file is for the multicast DNS implementation to discover instruments on MacOS. Likewise the avahi.c file is for Linux.

I suggest you place your cygwin/windows discover implementation in a separate file, e.g. cygwin_dnssd.c or similar and setup meson accordingly for when it builds under cygwin/windows.

psalic, 2024-09-28:
OK, will fix liblxi. 
lxi-tools windows branch is ready for pull request. You can check it

lundmar, 2024-09-28:
Please see https://github.com/psalic/lxi-tools/commit/08504c9f2d719a11b85d19a495d19ceb15770adb#commitcomment-147332434

psalic, 2024-09-28:
I've updated code on lxi-tools and liblxi.
In liblxi you will see one fix in bonjour.c
As I have problems during development and found later this bug
[https://github.com/lxi-tools/lxi-tools/issues/84](https://github.com/lxi-tools/lxi-tools/issues/84)

lundmar, 2024-09-28:
Please see https://github.com/psalic/lxi-tools/commit/64fc6d55d3ab10a0b6a189eca0d7dc2e01b22c8c#commitcomment-147333326

lundmar, 2024-09-28:
Please see https://github.com/psalic/liblxi/commit/b68fc9e539d8dbc9c73f4b0b9988bc0614b26ca4#commitcomment-147333530

The one fix you made in bonjour.c I would like you to add as a separate commit so that we can link it directly to the issue and close it.

psalic, 2024-09-28:
I missed double dns_sd.h check in case of cygwin, I loaded new code in liblxi.

lundmar, 2024-09-28:
FYI - it is actually easier if you just create the PR and then we fix stuff in review. Then any change you can just force push to same PR branch. It is a bit easier for the reviewer. Then I don't have to go peak in your repo.

psalic, 2024-09-28:
I open PR for both. we can continue there.

lundmar, 2024-09-29:
So now we have all the PRs merged and only the msys2 build repo is remaining. Good work.

Quick question, does all features work in the Windows version including the script feature?

lundmar, 2024-09-29:
Oh, don't forget to make a PR for the fix in bonjour.c

psalic, 2024-09-29:
@lundmar 

This task can be closed now. It was great pleasure to work with you.
I will now work on msys2_scripts on new repository.

I have few notes about lxi tools.

- in  scripts tab-> button "stop" somehow changed color. I  see this in debian trexie(testing) lxi-tools deb package.  It will also be in windows now. It looks some library changed something.

- in windows build fonts in instrument list and scpi tab response window are smaller.

psalic, 2024-09-29:
> So now we have all the PRs merged and only the msys2 build repo is remaining. Good work.
> 
> Quick question, does all features work in the Windows version including the script feature?

Yes, scripts are also working

lundmar, 2024-09-29:
> This task can be closed now. It was great pleasure to work with you.

You too. Thank you for all your contributions - good work! I hope you will stick around helping out with any Windows/Msys2 issues we may encounter in the future. Else the Windows support for lxi-tools will get in trouble since I don't use Windows myself and my Windows insight is limited. Of course, you are also welcome to help out with the project in general if you see any things that need fixing or new features that we should add. I aim to keep lxi-tools a useful little open source tool for LXI instruments.

> I will now work on msys2_scripts on new repository.

Great, consider it your playground. When you have something working perhaps we can give it a go at using the github CI service to build lxi-tools windows installers so that whenever I do a new lxi-tools release I can simply go fetch the windows installer from the CI service. If I recall correctly, the github CI service has various versions Windows VMs available for building.

> I have few notes about lxi tools.
> 
>     * in  scripts tab-> button "stop" somehow changed color. I  see this in debian trexie(testing) lxi-tools deb package.  It will also be in windows now. It looks some library changed something.

Right, I think what happened is that GTK4.x and libadwaita have progressed and a lot of API functions are marked as deprecated which is clear when you compile lxi-tools against newer GTk/libadwaita version. This could be directly related to this issue. When I get time I need to go through lxi-gui and replace the deprecated function calls with their new counterparts. I've put this in the TODO file.

The stop button is supposed to look bright red like this:
[image]
That color belongs to a specific GTK button color class if I remember correctly. It may have been changed or deprecated in recent version of GTK. The screenshot is from when running the snap version which is fixed on some older GTK4.x/libadwaita versions.

This is how it looks when compiled and run on latest Ubuntu 24.10:
[image]
Does not look good.

>     * in windows build fonts in instrument list and scpi tab response window are smaller.
Right, there is the same problem when running the Snap. I think it is a problem with not finding whatever font is specified to use in lxi-gui. In case of Windows, maybe the font needs to be shipped with the installer or a different font chosen.

psalic, 2024-09-29:
I forked windows-installer repo and will try to prepare github actions. It is first time i do this so it may take some time. Then I will release code to upstream repo.

About button. I am not sure if it is related to gtk. I made some test with older libraries and only updated gtk to latest version and button was still OK. I don't have good knowledge what could be wrong.

lundmar, 2024-09-29:
> I forked windows-installer repo and will try to prepare github actions. It is first time i do this so it may take some time. Then I will release code to upstream repo.

No hurry. In your time.

> About button. I am not sure if it is related to gtk. I made some test with older libraries and only updated gtk to latest version and button was still OK. I don't have good knowledge what could be wrong.

At least the button still works 😁

closed 2024-09-29 by lundmar

psalic, 2024-09-29:
yes, it works. In light theme is not so bad.

I push initial commit on my forked windows installer.
[https://github.com/psalic/lxi-tools-windows-installer](https://github.com/psalic/lxi-tools-windows-installer)

You can run with docker in linux
`sh docker/docker_command.sh`

structure of windows installed directory  will be in lxi-tools-redistributable/ and you can check how will look in the end

lundmar, 2024-09-29:
It would be good if you can describe such details in the README.md

psalic, 2024-09-29:
Before committing on upstream will be.