lxi-tools/lxi-tools #101: log example lua script error on Windows

The00, 2025-02-10:
Hello,

I've tried to use the log feature of the lxi-tools, but the supplied example fail with the following error : 

`[string "lxi-gui"]:50: bad argument #1 to 'close' (FILE* expected, got nil)`

I'm on Windows 11, and installed lxi-tools from the 2.8 installer.

Also, where a log file is supposed to be saved when using the function lxi_log_save_csv ? in the same folder as the script ?

Thanks a lot for this nice tool,

Bye

Théo

lundmar, 2025-02-10:
This script? https://github.com/lxi-tools/lxi-tools/blob/master/examples/lua/log.lua

It probably is a problem with where the log file is saved. By default it puts it where the application is started from.

Try specify the full path instead of just "log0.csv".

The00, 2025-02-10:
Yes this one. 
Okay, I will try. Should I use the Linux file path syntax or the windows one ?

lundmar, 2025-02-10:
I expect you will have to use the windows one.

The00, 2025-02-10:
It works ! 
thanks

closed 2025-02-10 by The00