lxi-tools/lxi-tools #102: Rigol MSO8000 screenshot handling

Scottapotamas, 2025-02-21:
When taking a screenshot with `lxi-gui` the error "Failure handling image format" is shown.

From the CLI I can see auto-detection uses the `rigol-1000z` plugin. The result is a .png file by name, but it's actually a bitmap as per Linux `file` output:

```
$ file screenshot_192.168.1.124_2025-02-21T18\:06\:53.png
screenshot_192.168.1.124_2025-02-21T18:06:53.png: PC bitmap, Windows 3.x format, 1024 x 600 x 24, image size 1843200, resolution 3937 x 3937 px/m, cbSize 1843254, bits offset 54
```

Manually specifying the `rigol-2000` plugin on the CLI correctly saves a .bmp file, and should be a better fit as the MSO8k is very similar to MSO5k.

https://github.com/lxi-tools/lxi-tools/blob/e1068ab38910adc96090bd557cf682ba9c254867/src/plugins/screenshot_rigol-2000.c#L101-L107

Appending `MSO8...` should be enough to support these units successfully, but I'm unsure if this wildcard syntax will correctly handle the even more niche MSO8000A units as well.

While we're here, the DS7000 and MSO7000 are also on the same software platform and should be treated the same way.

Happy to make a PR if needed.

___

For context, my `IDN?` looks like this

```
RIGOL TECHNOLOGIES,MSO8204,DS8A2xxxxxx77,00.02.02
```

lundmar, 2025-02-21:
@Scottapotamas Thanks for testing with your MSO8000, feel free to make a PR.

> Appending MSO8... should be enough to support these units successfully, but I'm unsure if this wildcard syntax will correctly handle the even more niche MSO8000A units as well.

Yes, I believe that should be enough to match the MSO8000A too.

Scottapotamas, 2025-02-21:
Thanks for the quick response and a great tool.

I'm trying to figure out the (very different) Rigol RSA3000/5000 behaviour now. If I get that working I'll just PR directly.

closed 2025-02-21 by Scottapotamas