lxi-tools/lxi-tools #104: it crashed running script

thewon86, 2025-03-12:
lxi-tools-2.8.exe

WIN10

Click "Script -> Open -> selcet example connect-vxi11.lua -> Run", the gui tool exit .
Click "Script -> Run", the gui tool exit .

lundmar, 2025-03-14:
I just tried do the same in the snap version on Linux. It does not crash.

It simply prints the following in the script output window pane:
```
Power Supply ID = -1.0
```
... because I don't have an instrument connected.

@psalic It is possible for you to see if you can confirm the issue on Windows?

lundmar, 2025-03-14:
@psalic PR merged. Thanks!

FYI - I've also pushed some fixes to liblxi to better handle invalid handles so it should not crash.

psalic, 2025-03-14:
@thewon86 
you can try new windows lxi tools.
[](https://github.com/lxi-tools/lxi-tools-windows-installer/releases/download/v2.8-4-g2a2dbe8/lxi-tools-2.8-4-g2a2dbe8.exe)

thewon86, 2025-03-21:
> [@thewon86](https://github.com/thewon86) you can try new windows lxi tools. https://github.com/lxi-tools/lxi-tools-windows-installer/releases/tag/v2.8-4-g2a2dbe8

when i open a example and run it, it seems ok.
but, just open the app, open file "connect-vxi11.lua" with notepad,  copy file context and paste into lxi-gui .  click "Run", it crash.

open the app, without inputting anything, click "Run",  it still crash

psalic, 2025-03-21:
Confirm that newly created not saved script crashes on windows.

lundmar, 2025-03-21:
@psalic Thanks for fixing the issue so quick!

psalic, 2025-03-21:
@thewon86 
Now it should work. New windows installer:
https://github.com/lxi-tools/lxi-tools-windows-installer/releases/download/v2.8-5-g482af29/lxi-tools-2.8-5-g482af29.exe

thewon86, 2025-03-25:
> [@thewon86](https://github.com/thewon86) Now it should work. New windows installer: https://github.com/lxi-tools/lxi-tools-windows-installer/releases/download/v2.8-5-g482af29/lxi-tools-2.8-5-g482af29.exe

Well done, it looks ok present.
Thanks.