lxi-tools/lxi-tools #107: Question: Labview device driver to LXI-TOOLS?

pir8radio, 2025-03-29:
Hey there, I have three devices I would like to control with LXI-TOOLS but you don't support them out of the box.  I was wondering, if you or someone with more knowledge than I, could convert these "labview" drivers into a device for lxi-tools?  I hope this is an easy ask.  2 & 3 say they support SCPI based text commands.  Thank you guys for considering.  :-)

1.) Rackmount power supply
Device: https://www.bkprecision.com/products/power-supplies/9115
LabView Driver: https://bkpmedia.s3.amazonaws.com/downloads/software/LV9115.zip

2.) DSI Signal Generator
Device: https://www.dsinstruments.com/product/rf-signal-generators/12ghz-usb-wideband-compact-signal-generator/
LabView Driver: https://www.dsinstruments.com/wp-content/uploads/2024/05/DSI-SIGGEN-Labview.zip

3.) DSI Digital Attenuator 
Device: https://www.dsinstruments.com/product/digital-attenuators/6ghz-90db-digital-rf-step-attenuator-usb-dat90/
LabView Driver: https://www.dsinstruments.com/wp-content/uploads/2024/05/DSI-DAT-Labview.zip

lundmar, 2025-03-29:
@pir8radio lxi-tools only work with test instruments that have ethernet (LXI = LAN eXtensions for Instrumentation). As far as I can tell the first instrument does not have ethernet and for the two other instruments ethernet is optional so you need to check if they have that option.

pir8radio, 2025-03-30:
Yes I have ethernet for the other two,  but most of my test instruments are serial over ethernet.

lundmar, 2025-03-30:
LXI covers various communications protocols that work over ethernet. I don't think your test instruments are lxi certified or compatible but there is a chance they will work with the RAW protocol connection option.