lxi-tools/lxi-tools #110: Siglent SDS1104X-E roll mode problem

xxxajk, 2025-04-07:
Siglent SDS1104X-E firmware 6.1.37R10

When in horizontal roll mode, the screen (and live copy) trigger a clear of the trace. 
Doesn't happen when using VNC.
Doesn't happen with webgui (although the webgui controls ends up oversized, and is unusable)

lundmar, 2025-04-07:
Which version of lxi-tools are you using?

Are you using the live capture feature of lxi-tools?

xxxajk, 2025-04-07:
Wow, that was quick!

lxi-gui 2.8 from snap

Yes, live capture.

Also noticing segfaults too, should probably grab a copy of the repo, compile and run valgrind on it....

xxxajk, 2025-04-07:
Yes, clears the trace on the instrument (ends up a very tiny flickering line segment)

xxxajk, 2025-04-07:
fwiw, grabbing a screenshot does this as well.

lundmar, 2025-04-07:
Yes. It is a different code path so it could be a bug in e.g. their vxi11 protocol or screenshot dumping mechanism somehow affecting the trace. All the screenshot plugin does is send the "scdp" command to the instrument to dump the screenshot (https://github.com/lxi-tools/lxi-tools/blob/master/src/plugins/screenshot_siglent-sds.c).

xxxajk, 2025-04-07:
I grabbed the next revision of firmware, and will see if that fixes it (it's a few revs higher). If not, then contact them?

lundmar, 2025-04-07:
I think so.

xxxajk, 2025-04-07:
No change, that's a bummer, guess I'll complain to them, and just not use it when in roll mode.

xxxajk, 2025-04-07:
Of course video via VNC is also a work-around...

lundmar, 2025-04-07:
I'm thinking it is probably the VXI11 protocol path that is disrupting whatever is going on on the instrument. VXI11 is known to be slow because of its overhead. But that is just speculation from my side.

lundmar, 2025-04-09:
It's an older scope so I'm not surprised if it is the effects of lack of processing power.

xxxajk, 2025-04-11:
Contacted Siglent with a pile of various bugs I've seen, they're forwarding the problems to R&D.
I'll keep you posted.