lxi-tools/lxi-tools #112: it don't append terminaton character behind command

thewon86, 2025-04-17:
Hi,
   In SCPI page, i select one command then click send, there is no termination character '\n'. Such as "*IDN?", it just send "*IDN?" but not "*IDN?\n"

thewon86, 2025-04-17:
as the same, the script removed last character '\n'
```
psu = lxi_connect("192.168.2.56", nil, nil, 2000, "VXI11")
respond = lxi_scpi(psu, "*IDN?\n")
print(respond)
lxi_disconnect(psu)
```

[image]

lundmar, 2025-04-17:
It depends which protocol you are using. In case of VXI11 there is no newline character added because the protocol itself takes care of termination.

In case of no protocol (RAW) a newline termination character is added.

lundmar, 2025-04-17:
As demonstrated in script RAW example:
https://github.com/lxi-tools/lxi-tools/blob/master/examples/lua/connect-raw.lua

lundmar, 2025-04-17:
Oh, I see that even in the VXI11 example script a newline is added.

https://github.com/lxi-tools/lxi-tools/blob/master/examples/lua/connect-vxi11.lua

I'm pretty sure that should not be needed protocolwise but it may depend on the instrument.

thewon86, 2025-04-17:
Maybe there is mistake. 
```
psu = lxi_connect("192.168.2.56", nil, nil, 2000, "VXI11")
respond = lxi_scpi(psu, "*IDN?\n")
print(respond)
lxi_disconnect(psu)
```
This is my lua script, it write command "*IDN?\n" with char '\n'.

[image]

But, in the VXI-11 Core Protocol packet, the char '\n' had disappeared

thewon86, 2025-04-17:
`respond = lxi_scpi(psu, "*IDN?\n6")`
If i add one more char, such as "*IDN?\n6" there.
will got

[image]

It's pretty sure that somewhere removed the last character '\n' in command string.

thewon86, 2025-04-17:
I'd test 2.7 in linux(Fedora), whick installed with dnf
and 2.8-5-g482af29 in Win10.
Thay are the some.

The second thing, how could i input '\n' into command line?

[image]

lundmar, 2025-04-17:
I've found the bug. It is fixed in fa239b380daff79d2661ed05aa5642d9d5ee14d9. Turns out isspace() strips more than just spaces! What a stupid function name.

lundmar, 2025-04-17:
> The second thing, how could i input '\n' into command line?

You can't. It is not meant to be a low level interface like that. Only when you press enter the newline is automatically added.

thewon86, 2025-04-17:
> > The second thing, how could i input '\n' into command line?
> 
> You can't. The gui is not meant to be a low level interface like that. Only when you press enter the newline is automatically added.

Yes，It could be one capability of lxi-gui, allowing the option to append either '\r', '\n', or '\r\n', or not append anything.  Like "keysight Interactive IO" dose, it can change EOL Sequence.

[image]

thewon86, 2025-04-17:
> Yes, we could add that to the preferences dialog. It should only apply to RAW though.

May VXI11 need it also. 
you known, when we click the button "search", lxi-gui invoke ["vxi11_discover"](https://github.com/lxi-tools/liblxi/blob/b14716ba12f296cb0d35640df5dcb84889fd978d/src/vxi11.c#L539), then [get_device_id](https://github.com/lxi-tools/liblxi/blob/b14716ba12f296cb0d35640df5dcb84889fd978d/src/vxi11.c#L354)  `length = vxi11_send(&data, ID_REQ_SCPI, strlen(ID_REQ_SCPI), timeout);` 
The `ID_REQ_SCPI` is defined "*IDN?\n"
For this reason, in my case, only "search" works. Send any script dose not gotting respond.

lundmar, 2025-04-17:
This needs to be investigated as I can't remember these details - it is a long time since I implemented this stuff. However, if I recall correctly VXI11 does not require line termination because the receiving instrument considers all received data a command to be handled. In fact, I expect LXI certified instruments may not accept line termination in VXI11 payload. This is not the case with TCP/RAW where there is no protocol involved so it needs the termination character to indicate to received that it can start process the received line.

What test instrument are you testing with?

lundmar, 2025-04-17:
With the latest changes this works with my LXI certified NGM202:
```
-------------------------------------
--  lxi-tools                      --
--    https://lxi-tools.github.io  --
-------------------------------------

-- Example: Retrieve ID from instrument (R&S NGM 202) via VXI11 protocol

-- Connect to power supply
psu = lxi_connect("192.168.0.107", nil, nil, 2000, "VXI11")

-- Print instrument ID
print("Power Supply ID = " .. lxi_scpi(psu,"*IDN?"))

-- Disconnect
lxi_disconnect(psu)
```

```
lundmar@wopr:~/projects/lxi-tools/lxi-tools/examples/lua$ lxi run ./connect-vxi11.lua 
Power Supply ID = Rohde&Schwarz,NGM202,3638.4472k03/101403,04.013
```

```
-------------------------------------
--  lxi-tools                      --
--    https://lxi-tools.github.io  --
-------------------------------------

-- Example: Retrieve ID from instrument (R&S NGM 202) via RAW (socket)

-- Connect to power supply
psu = lxi_connect("192.168.0.107", 5025, nil, 2000, "RAW")

-- Print instrument ID
print("Power Supply ID = " .. lxi_scpi(psu,"*IDN?"))

-- Disconnect
lxi_disconnect(psu)
```
```
lundmar@wopr:~/projects/lxi-tools/lxi-tools/examples/lua$ lxi run ./connect-raw.lua 
Power Supply ID = Rohde&Schwarz,NGM202,3638.4472k03/101403,04.013
```

thewon86, 2025-04-18:
As far as I know, whether an EOL (End of Line) is required or whether the EOL is '\r', '\n' or '\r\n' depends on the instrument's SCPI parser itself. Most instruments I have encountered use '\n', though some use '\r\n'. The data transmitted via TCP/RAW or VXI11 (i.e., the strings received and parsed by the instrument’s SCPI parser) should be identical. Therefore, regardless of whether TCP/RAW or VXI11 is used, it is more practical to let the user decide the EOL configuration.

I agree with what you said that "the receiving instrument considers all received data via VXI11 a command to be handled"

Further testing on the Agilent 33220A, sending "*IDN?\n" or "*IDN?" both works well.
I guess the EOL character is handled on the VXI11 server inside the instrument, and it works well regardless of whether EOL is appended to the command string.

lundmar, 2025-04-18:
In gui, maybe have separate settings for VXI and RAW and have default for VXI be none and default for be '\n'.